OpenStickCommunity/GP2040-CE #650: Xinput Player number add-on assigned static numbers not recognized by Windows 10

ElMeroMaquinero, 2023-12-01:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.5
* Name of device: Pi Pico
* Link to where you purchased the device: https://www.pishop.us/
* Is this a custom build?: no
* Operating System: Win10
* Browser: Google Chrome

### Expected Behavior

Please describe the behaviour you are expecting.
Using qty 4 pi pico in xinput mode.  Assigned player numbers 1-4 via web configurator.    

### Current Behavior

What is the current behaviour?
When PC is turned on/off the player numbers are mixed.  Reboot PC again and player numbers change.
Verified player numbers on  https://hardwaretester.com/gamepad

Checked web configurator and players numbers show correct assignment

Tried new fresh pi pico, no customizations only player number add-on but still not stable ID's.
Did a code nuke and re-install but same outcome

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Step 1
2. Step 2
3. ETC...

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

ElMeroMaquinero, 2024-09-16:
Wondering if anyone has done anything with this?  Not sure if i setup the issue correctly :/

mikepparks, 2024-09-16:
The issue is currently being investigated as it's not necessarily something that can be changed on the fly.  The order of the devices is determined by the OS, and how they're enumerated, so our method of forcing a player number is not foolproof.  That said, we're currently looking into a means of uniquely identifying devices to make them easier to assign. This may not solve the player number issue, but it may help with assigning specific devices to input configs.

fanminxiang, 2024-12-09:
The player number confusion occurs after the PC is restarted. Can we let player No. 2 delay 1 second to respond to the USB, and player No. 3 delay 2 seconds to respond to the USB? Perhaps this can solve the ID confusion problem.