OpenStickCommunity/GP2040-CE #662: Would you consider adding support of mouse

AlanZ-Git, 2023-12-09:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

I found it useful to set right-click in keyboard mapping. It helps a lot in using hitbox for Street Fighter 6.

Why I don't use XBOX mode to play SF6: I can get more keys to use with keyboard mode. And more keys help me to quickly change setting in training in game.

Why right-click is necessary: In SF6, "confirm" is A in XBOX, and F in keyboard. "Cancel" is B in XBOX, and right-click in keyboard (mode).

### Describe the solution you would like to see

Make it available to set right-click in keyboard mapping.

### Describe alternatives you've considered

not much.

### Can you submit a pull request?

Sorry I only know a little python.

### Are you planning on working on this feature yourself?

No. It would be great help it you do so. Thank you.

TheTrainGoes, 2023-12-10:
Hi @AlanZ-Git,

At this time I am not aware of anyone who is working on mouse inputs for HID-Keyboard mode.  

I will leave this open incase anyone sees it and would like to work on it.

AlanZ-Git, 2023-12-11:
Thank you! I will try to learn to read these code to see if I can make the new function.

AlanZ-Git, 2023-12-11:
@TheTrainGoes 
Thank you! I will try to learn to read these code to see if I can make the new function.

mikepparks, 2024-09-18:
Mouse buttons will be available from within the Keyboard Host, now renamed as Keyboard/Mouse host, with the release of version 0.7.10.

peterjc, 2025-03-18:
Can you use the "context menu" keyboard key usually bottom right between alt and ctrl, or must it actually be the mouse right-click?

TheTrainGoes, 2025-07-24:
Mouse inputs have since been added to the firmware and this will be closed.  

Note that this is not mouse movement, just the inputs.

closed 2025-07-24 by TheTrainGoes