OpenStickCommunity/GP2040-CE #667: Pin assigned to addon will not be unassigned

jack2game, 2023-12-16:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware - built from latest main branch
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: latest main
* Name of device: pico
* Link to where you purchased the device: x
* Is this a custom build?: No
* Operating System: Windows
* Browser: Chrome

### Expected Behavior

Pins previously assigned to addon should be automatically unassigned if unused, but this is not always the case.

### Current Behavior

Taking Turbo addon for example
- If PIN14 was previously assigned as Turbo pin
- Then in the addon settings, change PIN14 to PIN16, and save
- Both PIN14 and PIN16 will show as "Assigned to addon" in the pin settings page, even when PIN14 shouldn't be
- To unassign PIN14, you need to change the Turbo pin back to PIN14, save, then change to -1, save again

### Steps to Reproduce

As explained above.

### Screenshots & Files

x

jack2game, 2023-12-17:
Just want to add the comment above only works for Profile 1 (Base Profile):
"To unassign PIN14, you need to change the Turbo pin back to PIN14, save, then change to -1, save again"

For profile 2-4, there's actually no way to unassigned the previously assigned PINs.

Similarly, if any new PIN is assigned to a Addon, it only appears on Base Profile, but not Profile 2-4.

bsstephan, 2023-12-18:
Two things, though in general I believe all of this to be true.

>     * If PIN14 was previously assigned as Turbo pin
> 
>     * Then in the addon settings, change PIN14 to PIN16, and save
> 
>     * Both PIN14 and PIN16 will show as "Assigned to addon" in the Pin Mapping page, even when PIN14 shouldn't be
> 
>     * To unassign PIN14, you need to change the Turbo pin back to PIN14, save, then change to -1, save again

I have some code that attempts to unassign the old pin, but it was overzealous is some form so I think it may not be triggering here when I intend it to. I can try to take a look.

> Just want to add the comment above only works for Profile 1 (Base Profile): "To unassign PIN14, you need to change the Turbo pin back to PIN14, save, then change to -1, save again"
>
> For profile 2-4, [...]

This is a tricky one. I think this probably needs a UI/UX conversation, because the worst part is the "Assigned to Addon" pins in profiles 2-4 have no affect anyway, because profiles aren't allowed to unset the pins of addons at present. So, it's almost as if they shouldn't be restricted in profile 2-4, but what's probably better is they shouldn't even be *reflected* in those profiles' UI since they don't matter.

bsstephan, 2023-12-27:
#726 will address the latter problem where the assignments don't affect the profiles. It wasn't that tricky in the end. The former problem is trickier and needs another look, but the workaround above (assigning to -1 manually) works for the interim.

PR #726 (2023-12-27): v0.7.6-RC3 Fix: manage ASSIGNED_TO_ADDON profile pins as well as core pins
`docToPin` and our USB D+/D- handling code only managed the core profile configuration, but it also needs to set the fields in profiles.

fixes #667
commit 6f1af543 (2023-12-27): v0.7.6-RC3 Fix: manage ASSIGNED_TO_ADDON profile pins as well as core pins (#726)

manage ASSIGNED_TO_ADDON profile pins as well as core pins

docToPin and our USB D+/D- handling code only managed the core profile
configuration, but it also needs to set the fields in profiles.

fixes #667
closed 2023-12-27 by arntsonl