OpenStickCommunity/GP2040-CE #670: Abnormal Focus Mode Behaviour

BolSadguy, 2023-12-17:
### Prerequisites

Please check the following before posting an issue / bug report.

- [Yes ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [Yes ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [Yes ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [Yes ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: v0.7.5
* Name of device: RP2040ABB
* Link to where you purchased the device: Direct from TheTrain
* Is this a custom build?: No
* Operating System: Windows
* Browser: Edge

### Expected Behavior

Please describe the behaviour you are expecting.

Focus Mode designated pin shorted to GND:
- OLED turned off
- RGB lights turned off

### Current Behavior

What is the current behaviour?

Focus Mode designated pin shorted to GND:
- OLED remains on but is frozen
- RGB lights remain on and still react to inputs

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

Webconfig: Set Focus Mode to GPIO 26 (Option 3), enable Lock OLED and Lock RGB
Connect SPST switch to GPIO26 and GND

### Screenshots & Files

[image]

TheTrainGoes, 2023-12-21:
@BolSadguy can you test this on the v0.7.6 RC1 if you have a chance?

TheTrainGoes, 2023-12-21:
Tested in v0.7.6 RC1 and this is still an issue

BolSadguy, 2023-12-21:
> @BolSadguy can you test this on the v0.7.6 RC1 if you have a chance?

Just tested it, the behaviour is unchanged from what was reported.

TheTrainGoes, 2023-12-21:
Testing Focus mode as per https://github.com/OpenStickCommunity/GP2040-CE/issues/670 in v0.7.6 RC1 and the locking of the OLED and RGB is not working as intended.

Steps to reproduce:
1 - Nuke Haute42 T16
2 - Load v0.7.6 RC1
3 - Web-config - turn turbo off and replace with focus mode set to lock OLED and RGB + buttons
4 - Restart

Expected behaviour:
When holding the focus mode button down the OLED should be off, the RGB should stop and the specified buttons should not work

Actual behaviour:
When holding the focus mode button down the OLED remains on but no longer registeres inputs, the RGB continues to run but no longer registers inputs.
The specified buttons no longer work (they are working as intended)

TheTrainGoes, 2024-02-04:
For the time being we have removed the ability to turn off the dispaly and RGB from focus mode.  Closing this request, may revisit in the future.

closed 2024-02-04 by TheTrainGoes