OpenStickCommunity/GP2040-CE #693: An input command bug

yercn, 2023-12-22:
I'd like to report an issue. When I'm playing the game "King of Fighters," there is a problem with the input command →↘↓↙←→+A/C . When my character is in the 1P position,  it's extremely difficult to execute this command accurately. However, when the character is in the 2P position (the opposite side), the same command works fine. I hope this issue can be fixed.

InfraredAces, 2023-12-22:
This sounds like an execution error and not an issue with the firmware itself. You will need to practice the inputs more and become more consistent with them..

yercn, 2023-12-22:
No, I don't think so. I use keyboard as a gamepad to play the game for more than 10 years. When I use keyboard and other gamepad controller (keyboard style too) to input the command, all of them works fine. And I tried to use other gamepad firmwares of RP2040, such as https://github.com/oomek/Pico-Dual-Gamepad and so on, they works fine too. I'm favor with GP2040-CE, so I expect for the perfect version of it. Pls check issue in the The King of Fighers, then you'll find that out.

yercn, 2023-12-22:
Here is the problem. When you in the 1P position, If you hold the forward key(→) and do not release it, and then input the ↘↓↙←→+A/C, the command →↘↓↙←→+A/C will not be excuted. You had to release the forward key(→) first, and the input the →↘↓↙←→+A/C, in this situation, the command works. BUT, it is not right. This will delay the attack in the game. HOWEVER, when you in the 2P position, you can easily holding the forward key(←) and then input the remaining command ↙↓↘→←+A/C, the input will works perfect just like other keyboards and gamepads.

TheTrainGoes, 2023-12-22:
What are your SOCD settings? 

Are you able to provide a video of the issue happening as well as a recording of the onscreen inputs?

TheTrainGoes, 2023-12-22:
@yercn reading back on this I am fairly certain this is an input issue and SOCD catching it.

You said above that you use a keyboard (which is likely not to have SOCD built into it).  

The `←→+A/C` and `→←+A/C` parts of this will both trigger SOCD rules if the directions are not inputted independently for GP2040-CE.  

Presuming that you are inputting the `←→` or `→←` at the same time on your keyboard the game is simply resolving this in your favour and executing the combo.

If you are playing on a computer you may want to look at disabling the SOCD (setting to none) and see if you are able to perform the moves as expected.  If this works than it is an input issue.

yercn, 2023-12-22:
Yes, you are right. I've tried it, and the problem is solved now.  Thank you very much

TheTrainGoes, 2023-12-22:
Great to hear - will close this off.

closed 2023-12-22 by TheTrainGoes