OpenStickCommunity/GP2040-CE #694: Debounce slider reconsideration

NielsenDias, 2023-12-22:
I just migrated from a dedicated arcade board to gp2040 and I am getting problems with double presses on a couple of buttons (lever-less box). They work perfectly when connected to the previous board.

And after troubleshooting a bit narrowed it to debouncing. For now I'll just replace the bouncy buttons to solve the problem, but, I think it'd be awesome to have a debouncing slider (already mentioned in #49 ), even if it possibly could introduce some latency at the cost of no double presses.

TheTrainGoes, 2023-12-22:
Hi @NielsenDias ,

We have debounce added into v0.7.6 RC1 if you want to check it out.

NielsenDias, 2023-12-22:
I downloaded the pre-release and tested out a bit. The double presses on click are gone, awesome work there!
But... the problem now is on release :joy:, press and holding correctly triggers, but, when I release the button it triggers another press.

Downgraded to v0.7.5 and exchanged the buttons for now.

TheTrainGoes, 2023-12-22:
Which buttons are you using @NielsenDias ?  KS by chance?

NielsenDias, 2023-12-22:
> Which buttons are you using @NielsenDias ? KS by chance?

Old ass sanwas..

TheTrainGoes, 2023-12-22:
Very odd.  I might see if you can replicate this on newer switches

dogtopus, 2024-01-18:
Shitty rubber domes also have the same issue. I learned it the hard way when I was modding a cheap toy into a game controller with GP2040.

I'd suggest adding optional debouncing on release as well, with a clear warning that enabling it might introduce unwanted latency when mashing the buttons under the debouncing threshold.

dogtopus, 2024-01-18:
@NickGuyver What I mean is GP2040 currently uses the "first press only" algorithm with a timeout, but has no such handling on release. This is somewhat expected because otherwise it could delay a legit press when debouncing threshold was set too high.

b3sky, 2024-01-21:
Sometimes a double click occurs when the button is released. 
tested it on both pico boards and compatible boards.

closed 2024-02-04 by TheTrainGoes

dogtopus, 2024-02-04:
@TheTrainGoes I don't think the double press on release is fixed yet. The code still does no debounce on "release then press" situation.