OpenStickCommunity/GP2040-CE #704: Bug for macro failure when setting multiple macros

wowplayhitbox, 2023-12-25:
* Firmware Version:[v0.7.6-RC2](https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.6-RC2)
* Name of device:GP2040-CE_0.7.6_Haute42.uf2
* Browser:chrome

When setting multiple macros, the triggering of the macro will fail. It will be valid for a while and invalid for a while. When only one macro is set, the triggering of the macro is completely normal.

http://192.168.7.1/macro
[image]

https://github.com/OpenStickCommunity/GP2040-CE/assets/154765649/b8e099e2-42cd-4b86-be67-5e86932904c2

fvolpe83, 2024-01-17:
same here, tested both with macro button pin and without:
[image]
[image]
as soon as the second macro is triggered, the first one becomes unavailable in both cases

fvolpe83, 2024-01-17:
added a third macro and nullifies the other two, looks like once the latest one gets used once, the others can't

alvarodiegomaia, 2024-01-17:
I got the same problem and posted it in the haute42 discord.

somerandomguy09, 2024-02-11:
I'm also afflicted by the same bug; as [fvolpe83] says. The moment the newest macro is triggered all previously assigned macros stop working. 

I'm on firmware 0.77.  Device is Haute42 T16. 

I've tried playing with the interruptible and exclusive settings with no luck either. For now I'm stuck only utilizing one macro at a time

WalkWithBlade, 2024-03-02:
I got the same problem!

InfraredAces, 2024-03-08:
Interesting. Given 2 macros, A and B as configured in the image below. 

If you press A, then pressing B won't work. However, if you press A while holding B, A will be triggered, then B will be triggered and B will be the only macro to work. You can do the reverse to change A back to being the only working macro.

If you press A and B rapidly (either alternating or one multiple times then the other), both will work, but if you stop, the last one you pressed will be the only one that works. Until you do any of the above listed steps.

[image]

TheMasterPrawn, 2024-03-15:
Same problem here. The interface is kinda confusing. But I started to get teh hang of it. It will only ever run the first Macro for me though. 
[image]

Surely there must be more people having this problem ? Do we revert firmware to fix for now ?

TheTrainGoes, 2024-03-28:
This has been addressed in v0.7.8

wowplayhitbox, 2024-04-03:
This issue has been fixed by new version. I downloaded the version here and verified that it is completely normal.
https://github.com/OpenStickCommunity/GP2040-CE/actions/runs/8501274945#artifacts

TheTrainGoes, 2024-06-11:
Closing this issue - resolved

closed 2024-06-11 by TheTrainGoes