OpenStickCommunity/GP2040-CE #707: SH1106 OLED screen does not function on Waveshare RP2040-Zero on 0.7.6-RC2

TheOpponent, 2023-12-26:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.6-RC2
* Name of device: Waveshare RP2040-Zero, DIYmalls 1.3" inch OLED Display Screen Module I2C 4pin 128x64 SH - 1106 3.3V-5V IIC Address 0x3C White for Arduino U NO Mega-2560 Nano ESP8266 NodeMCU ESP32
* Link to where you purchased the device: https://www.amazon.com/dp/B0BZ8D4CJM, https://www.amazon.com/dp/B0CCL88VTG
* Is this a custom build?: Yes
* Operating System: Windows 10
* Browser: Vivaldi 6.4.3160.44

### Expected Behavior

Please describe the behaviour you are expecting.
The OLED screen functions as it does in 0.7.5, with the screen configured to block i2c1 with SDA and SCL set to pins 26 and 27.

### Current Behavior

What is the current behaviour?
The screen does not turn on in 0.7.6-RC1 or 0.7.6-RC2.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Connect the SH1106 screen to pins 26 and 27.
2. Install latest 0.7.6 RC to Waveshare RP2040-Zero and configure the peripheral settings to use pins 26 and 27.
3. Reboot and check the screen.
4. Reverting to 0.7.5 corrects the issue with the screen without a config reset or flash-nuke necessary.

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

n/a

jakubkoziol, 2023-12-28:
What I found after running into the same problem and running some tests:

Using the same i2c block as what is specified in the board config under I2C_BLOCK works fine.
Using the other i2c block appears to cause problems.

With Pico as an example, running the default uf2, OLED will work when connected to 0, 1.
Changing pins to 4, 5 will also work.
But changing pins to 2, 3 won't.

Running a uf2 where the only change is I2C_BLOCK set to i2c1 and OLED pins set to say 26, 27 OLED will work.
Changing pins to 2, 3 will also work.
But changing pins to 0, 1 won't.

TheOpponent, 2023-12-28:
The screen appears to be working in 0.7.6 final, though a minor bug remains when upgrading directly from 0.7.5: If the I2C peripheral was enabled in 0.7.5, the pins for i2c0 and i2c1 are marked as reserved by addon in the Pin Mapping page, even if their respective i2c block is disabled. To fix this, it is necessary to temporarily enable the block in question, unset the pins, and then disable the block.

TheOpponent, 2024-01-01:
After building another controller with Waveshare RP2040-Zero, I've discovered that there's an issue with the default i2c0 mappings. Because i2c0 is mapped to pins 0 and 1 by default, they conflict with the default mappings for S1 and S2, which can prevent them from being recognized and therefore locking out the web config. In order to use the screen with i2c1, it is necessary to first enable i2c0, reboot to the web config, unset the i2c0 pins, and then enable i2c1 and ensure S1 and S2 are mapped.

TheTrainGoes, 2024-02-04:
This should be resolved now after the config cleanup

closed 2024-02-04 by TheTrainGoes