OpenStickCommunity/GP2040-CE #725: Discussion/POC: simplify Dual Directional Input by removing some/most combination modes

bsstephan, 2023-12-27:
With 0.7.6 getting multiple pins outputting the same direction button, and #724 potentially allowing one pin to output multiple buttons, I think we should talk a bit about the Dual Directional Input addon. Specifically, I think at least one of the four combine modes is unnecessary now, and perhaps three of them (all but None) are on the chopping block.

## Notes

### Combination Mode Mixed

> Combines both the Gamepad directions and Dual Direction Input directions and allows for all 3 SOCD modes.

This essentially has DDI work as 4 extra directional buttons that contribute to the core gamepad output. I think this is no different than mapping those buttons as more core directions, come 0.7.6, so I think this mode can be removed. Note that "Dual D-Pad Mode" has no relevance to this mode.

### Combination Mode Gamepad

> Gamepad always takes over when pressed, otherwise Gamepad and Dual act independently.

I have never fully understood this feature. The Dual "half" of the output is configured to be LS/DP/RS specifically, independent of the core gamepad's setting. Take the following situation:

* core Gamepad in DP mode
* DDI Mode in RS mode
* DDI combination mode "Gamepad"

If you...

1. press Gamepad Up, you get DP Up *and* RS Up (this is probably a bug, but I don't know since when. I'm on v0.7.6-RC1)
2. press DDI Up, you get RS Up (okay)
3. press Gamepad Up and DDI Up, you get DP Up and RS Up (this is probably the intentional behavior, unlike the first situation)
4. press DDI Down, you get RS Down (obv.)
5. press DDI Down and Gamepad Up, you get DP Up *and RS Up* (unsure about which of this is desired, but this is Gamepad taking priority over DDI, and forcing DDI's Down to be core's Up, as described by the Combination Mode)

I think this is *supposed* to work like Dual Directional mode below, but right now it is way buggy...

### Combination Mode Dual Directional

> Dual always takes over when pressed, otherwise Gamepad and Dual act independently.

I don't understand the use case for this feature either, but it works more like how I thought it would.

* core Gamepad in DP mode
* DDI combination mode "Dual Directional"
* (note that Dual D-Pad Mode again doesn't matter here)

If you...

1. press Gamepad Up, you get DP Up
2. press DDI Up, you get DP Up
3. press Gamepad Up and DDI Down, you get DP Down (order does not matter)
4. press Gamepad Left and DDI Left, you get DP Left (obv.)
5. press Gamepad Left and DDI Right, you get DP Right (order does not matter)

This is more clearly the DDI inputs taking precedence over the core gamepad inputs. However, I'm not sure what the use case for the behavior is. Again with multiple pin to button mappings possible now, I'm not sure how this precedence is any different than First Input Wins SOCD or Second Input Wins SOCD depending on your desired form of buttons.

### Combination Mode None

> Gamepad input and dual directional act independently of each other.

This does exactly what it says, and is maybe the only mode that *intentionally* uses the Dual D-Pad Mode setting. Core gamepad and DDI directions act independently, each getting SOCD cleaned independently (unless they're both contributing to the same output mode, in which case they get SOCD cleaned as one). This essentially allows for an extra set of directionals that *can* be mapped to something different than the core gamepad, distinguishing this mode (and maybe only this mode) from the ability to map multiple pins to the same direction in v0.7.6.

## Conclusion

My opinions below, but I intend to act on this, so discussion welcome before I go gutting code:

* Mixed Combination Mode: remove the mode, it's obsolete. Migrate mixed users to more non-DDI directional mappings.
* Dual Directional Combination Mode: remove the mode, it seems like a more complicated way to achieve something achievable in core, absent someone (pinging @arntsonl) explaining the use case for this better than I can understand it.
* Gamepad Combination Mode: remove the mode, as per Dual Directional, or else if it's important, fix it.
* None Combination Mode: keep it, this has a unique feature beyond what's possible in core, at least until we refactor things further.
* Dual D-Pad Modes: keep them all, they are relevant to None Combination Mode.

I think removing the three combination modes above will greatly simplify the addon, which is pretty hard to read, and further the ability for us to fold DDI into core in some fashion in the future. Comments desired.

TheOpponent, 2023-12-28:
I think this discussion can be combined with the Tilt Input addon discussion in #624, since the Tilt Left Analog and Tilt Right Analog direction pins functionally act as extra inputs for left analog and right analog at 100%, the same as Combination Mode None in this addon. The tilt factor settings do not appear to have any bearing on the two tilt direction pins, which to me says that the extra analog pins aren't actual tilts in the traditional sense but rather additional full direction inputs.

bsstephan, 2024-01-06:
If the tilts don't actually tilt, I think that should be filed as a separate issue, because the whole alleged purpose of that addon was for them to tilt (rather than go to the extreme). I think DDI is conceptually separate as a simpler way to configure digital outputs (including DP).

bsstephan, 2024-03-17:
Updates:

Further testing has led to the conclusion that if nothing else, Mixed mode allows for two layers of Last Input Wins SOCD to be applied on top of both directionals (since they have distinct SOCD cleaning histories and phases), which is interesting enough that it survives.

None combination mode is unique for simultaneous different direction output modes, and survives.

I suspect Gamepad and DDI combination modes will be unique once fixed/drilled into, so this might become more of a docs and tech debt update than anything else.

bsstephan, 2024-03-17:
## DDI Mixed, Last Input Wins SOCD

1. Core Left = Left
2. Core Left + Core Right = Right
3. Core Left + Core Right + DDI Left = Left
4. Core Left + Core Right + DDI Left + DDI Right = Right

3 and 4 are not possible if Mixed mode were removed and just replaced with more core directions. This makes sense because they're no different to the firmware. Hence, Mixed mode stays.

## DDI Mixed, First Input Wins SOCD

1. Core Left = Left
2. Core Left + Core Right = Left
3. Core Left + Core Right + DDI Left = Left
4. Core Left + Core Right + DDI Left + DDI Right = Left

First input behavior is maintained still.

Gamepad and DDI combination modes still as described above. Gamepad mode still seems buggy to me, and DDI seems to work as described.

bsstephan, 2024-03-17:
In testing DDI combination mode, I have confirmed the behavior of #823.

bsstephan, 2024-03-24:
In the end, all the modes are staying, just the code is easier to read and the Gamepad Override mode should be fixed now.