OpenStickCommunity/GP2040-CE #731: Display no longer turns on when booting into web configurator after update

Soundtoxin, 2023-12-28:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: GP2040-CE 0.7.6
* Name of device: Haute42 G16
* Is this a custom build?: No
* Operating System: Guix System GNU/Linux
* Browser: Qutebrowser

### Expected Behavior

Previously (before updating, 5 minutes ago) if I held start on plug the screen would come on and display some info like the local IP to go to for the web configurator and I believe also said the firmware version.

### Current Behavior

I get an orange power LED and am able to pull up the web configurator fine but the little screen is dark now in this mode.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Update to latest 0.7.6 stable
2. Hold start on plugin
3. Watch the display/lights for activity

I do remember the changelog saying something about having to set up boards again due to breaking changes, but my custom keyboard mode keybinds are still there so it's not like I lost all my settings. I did also go into the display configuration section of the web configurator and it was set to enabled already. Plus it works in other modes.

TheTrainGoes, 2023-12-28:
Hi @Soundtoxin 

As per the notes about this release please nuke the device first and then load the firmware on.  This will resolve the issue but you will need to re-do any custom mappings.

Soundtoxin, 2023-12-28:
Thank you. That indeed fixed it. I used flash_nuke to be safe, but would resetting setting in the web configurator have worked as well for a less extreme solution?

closed 2023-12-28 by Soundtoxin

TheTrainGoes, 2023-12-28:
Hi @Soundtoxin ,

I do not believe so as we changed how a number of things work on the back end.