OpenStickCommunity/GP2040-CE #749: Xbox Series X Automatically Turns Back On After Soft Power Down

arobotbitme, 2023-12-31:
**Steps/issue**:
1. Soft shutdown of Xbox Series X using GP2040 controller (details below) by holding Home/Guide and selecting the shutdown option. 
2. System powers down 
3. System immediately automatically boots back up as long as GP2040 controller remains plugged in

**Troubleshooting**:
-Tried disconnecting the controller by pulling the USB cable out while the system is shutting down. System does not turn back on when controller is not present during shutdown--this is my workaround for now.

-If I plug the controller back in after waiting about a minute, it will **not** automatically turn the Xbox Series X back on; issue only seems to happen if the controller is still connected right at the end of the system's soft shutdown process. 

-No similar issue on PS4. I feel like this could be a problem with the Magic X specifically? 

-All other functionality normal across all tested platforms including Series X 

**Firmware and Hardware Details:**
Xbox Series X
GP2040, 0.76 FW
Fightbox
Raspberry Pi Pico
Homemade passthrough "board" using the female end of a USB extender and 22ohm resistors (hasn't given me a problem before) 
Magic X Wireless Converter/Dongle connected to GP2040 pass-through
Pass-through D+ on pin 14, D- on 15, powered by VBUS
Both PS4 Pass through and Xbox One Pass through options turned on in web configurator

TheTrainGoes, 2024-01-06:
Hi @arobotbitme ,

Thank you for the report.  We will take a look into this but the solution may just be to unplug the controller while turning the system off and plug it back in after the ssytem has been turned back on.

arntsonl, 2024-06-03:
I think 0.7.9 will fix this :O

TheTrainGoes, 2024-06-21:
This issue is fixed as of v0.7.9

closed 2024-06-21 by TheTrainGoes