OpenStickCommunity/GP2040-CE #764: Keyboard Host - Up and Down are reversed in DPad Digital Mode (Stock .uf2)

InfraredAces, 2024-01-03:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ X ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ X ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ x ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ X ] I am reporting the issue to the correct repository

### Context

* Firmware Version: 0.7.6
* Name of device: Keyboard Converter
* Link to where you purchased the device: Built from hardware repo files
* Is this a custom build?: No
* Operating System: Linux Arch
* Browser: Chrome

### Expected Behavior

- Pressing the keyboard key mapped to the Up input results in an Up input. 
- Pressing the keyboard key mapped to the Down input results in an Down input. 

### Current Behavior

Without changing any settings in the firmware and Keyboard Host add-on, the inputs for Up and Down are reversed even though the Web Configurator indicates that they are correct.

- Pressing the keyboard key mapped to the Up input results in an Down input. 
- Pressing the keyboard key mapped to the Down input results in an Up input. 

Note: This bug only occurs when the gamepad is set in DPad Digital Mode. DPad Left Analog and DPad Right Analog both have the Up and Down inputs correctly mapped

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Flash Nuke board
2. Flash [GP2040-CE 0.7.6 firmware](https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.6/GP2040-CE_0.7.6_KeyboardConverter.uf2) for the Keyboard Converter from download site
3. Press Up Arrow in DPad Digital mode, get Down input
4. Press Down Arrow in DPad Digital mode, get Up input

This also occurs if you change the keyboard mappings under Keyboard Host add-on settings to some other set of keyboard keys.

If you change the DPad mode to Left Analog or Right Analog, the directions are input correctly. When you switch back to DPad Digital, you see the above behavior once more.

### Screenshots & Files

Keyboard Host Settings:

[image]

Pressing Up Arrow Key:

[image]

Pressing Down Arrow Key:

[image]

TheTrainGoes, 2024-01-06:
This has been fixed and is awaiting merge.

mikepparks, 2024-01-08:
Merge completed.

closed 2024-02-04 by TheTrainGoes