OpenStickCommunity/GP2040-CE #779: Can an analog trigger be added?

ydegle, 2024-01-05:


ydegle, 2024-01-05:
> _No description provided._

ydegle, 2024-01-05:
Can an analog trigger be added?It will be more perfect. Also, the ADC used should not be too expensive.

TheTrainGoes, 2024-01-06:
Hi @ydegle ,

An analog trigger just uses one axis.  

In o ur current implementation you can have one analog stick and two analog triggers if your setup allows for it.  You will need a device that has all four ADC pins broken out though as the Pico only has three broken out by default.  

Since the RP2040 only has 4 ADC pins we are limited to either 2x analog stick or 1x analog stick and 2x analog triggers (or 4x analog triggers if there is a use case for that).

Anything further will require an expansion module which is being looked into but we have no further information at this time.

ydegle, 2024-01-06:
Dear friends:
       thank you very much for your reply! I wanted to use it for racing game controls, but there was no way to get the throttle and brakes, which are usually simulated triggers. This is very confusing to me.

---- Replied Message ----
| From | ***@***.***> |
| Date | 01/07/2024 02:03 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [OpenStickCommunity/GP2040-CE] Can an analog trigger be added? (Issue #779) |

Hi @ydegle ,

An analog trigger just uses one axis.

In o ur current implementation you can have one analog stick and two analog triggers if your setup allows for it. You will need a device that has all four ADC pins broken out though as the Pico only has three broken out by default.

Since the RP2040 only has 4 ADC pins we are limited to either 2x analog stick or 1x analog stick and 2x analog triggers (or 4x analog triggers if there is a use case for that).

Anything further will require an expansion module which is being looked into but we have no further information at this time.

TheTrainGoes, 2024-02-04:
Until we have an I2C expander done this will be closed as not planned.

closed 2024-02-04 by TheTrainGoes

ydegle, 2024-06-02:
Dear all:
    Thank you very much for your reply!
    How do I get custom firmware? Such as D-input device name, PID,VID, etc. I can make a donation or pay for it.
    

|
fu
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 1/7/2024 02:03 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [OpenStickCommunity/GP2040-CE] Can an analog trigger be added? (Issue #779) |

Hi @ydegle ,

An analog trigger just uses one axis.

In o ur current implementation you can have one analog stick and two analog triggers if your setup allows for it. You will need a device that has all four ADC pins broken out though as the Pico only has three broken out by default.

Since the RP2040 only has 4 ADC pins we are limited to either 2x analog stick or 1x analog stick and 2x analog triggers (or 4x analog triggers if there is a use case for that).

Anything further will require an expansion module which is being looked into but we have no further information at this time.