OpenStickCommunity/GP2040-CE #781: Allow disabling SOCD only for keyboard mode

Soundtoxin, 2024-01-05:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

I need* to have SOCD cleaning disabled for keyboard mode to properly play rhythm games like vivid/statis, osu!mania, quaver, etc. However, the SOCD mode is currently global and also affects xinput. There's no shortcut bound by default to turn SOCD off, just back on with one of the modes, so I tend to leave it off all the time for convenience. This could pose a problem when playing certain PC fighting games, I imagine (though not sure how common that is, I hear many games implement their own SOCD cleaning).

*Basically I'm binding the buttons that are normally the directions to keyboard letters, e.g. S, D, and F for left, down, and right. These games will have jumps/chords/double-notes and so on where you need to hit multiple of the note buttons together, sometimes all of them. With SOCD cleaning enabled you'll drop hold notes if you hit left and then right even if your left and right buttons are really just pressing some letters. This can also be tested with SOCD on/off using [QMK's keyboard layout tester](https://config.qmk.fm/#/test) since it highlights keys you're currently pressing.

### Describe the solution you would like to see

Either never apply SOCD cleaning to keyboard mode (is there actually a use-case for having it?) or allow us to separately set SOCD for xinput and keyboard mode (and any other modes it may affect, though I think those are the only ones currently).

### Describe alternatives you've considered

Kinda already mentioned two above but if not letting us set SOCD per-mode, just disabling it in keyboard mode by default somehow would be fine as well.

### Can you submit a pull request?

No.

### Are you planning on working on this feature yourself?

No.

Sidenote: The docs for [SOCD Cleaning Off](https://gp2040-ce.info/hotkeys#socd-cleaning-off) currently say:
>**This cleaning mode is only available in XInput mode** as PS3/DirectInput, PS4 and Nintendo Switch modes do not support setting SOCD Cleaning to Off and will default to Neutral SOCD Cleaning mode.

Emphasis mine. In other words it doesn't even mention that SOCD cleaning (or at least the setting to disable it) affects keyboard mode. Perhaps it was unintended that it was left on. I'm guessing keyboard mode is the most niche mode among users of this firmware.

bsstephan, 2024-01-06:
SOCD mode will be applied to keyboard mode as people play fighting games, shmups, etc. in keyboard mode sometimes too. I don't think there's a high likelihood of tying SOCD mode to input mode, as it seems more confusing than not; have you tried setting SOCD modes to different hotkeys or a slider and using those to switch when you need to?

Soundtoxin, 2024-01-06:
> have you tried setting SOCD modes to different hotkeys or a slider and using those to switch when you need to?

Haven't got a slider, that'd be nice I guess. Seems like GP2040-CE controllers keep wanting to get bigger to make use of all these new features. I've got a bunch of different ones already. Getting to be an expensive journey.

I haven't messed with certain features like profiles or messed with the default shortcuts too much as I'm not sure what all key combos are still open/unbound or if I want to overwrite one of the default binds. I probably won't ever want to use the shortcut to swap up/down, but being used to the default bindings has advantages when switching controllers or helping other people with their controllers. I'll consider adding a bind to my config to disable SOCD cleaning to make going back and forth on the fly possible if I really have no other option, but having to hit that all the time (and more importantly *remembering* to hit that) is not ideal at all. In a tournament setting of course I would double-check these things to stay out of trouble, but playing solo/online or with friends I don't even think about SOCD typically.

I could see separate SOCD per-mode being confusing without a screen on your controller (and only some of mine have screens), but on ones with a screen it shows your current setting, so you can get a quick sanity check from that. Maybe we could have an optional toggle somewhere in the web configurator to separate them / override the setting for keyboard mode then and leave the defaults as they are. I didn't realize some people played fighting games in keyboard mode. I guess maybe that's easier in some emulators or fightcade. 

I'm starting to wonder which games SOCD cleaning even actually matters in (excluding the console games since we're talking xinput and keyboard modes) and if I should stop worrying about having it off. In a SHMUP what would happen with it off? I'm imagining pressing left and right either has the last movement button win out or maybe you'd be stuck in place not moving. Real keyboards wouldn't have SOCD cleaning, so can we really even say it's an accurate keyboard reproduction? Correct me if I'm wrong but in any game with keyboard controls they would implement SOCD stuff on the game side and so it would react the same to a real keyboard or a GP2040-CE controller in keyboard mode with SOCD cleaning disabled.

TheTrainGoes, 2024-01-12:
Hi @Soundtoxin ,

This is a difficult one to answer since on a computer different games will react differently to SOCD with no SOCD filtering.  

To summarize where we are at with this:
1 - SOCD will not be removed for keyboard mode as a whole since it is heavily used in games that need it (SF6 etc.)
2 - Setting the device to keyboard mode and SOCD off should address your concerns, if you are still seeing SOCD rules being implemented then it will be the game / emulator / other settings doing so

Soundtoxin, 2024-01-12:
> To summarize where we are at with this: 1 - SOCD will not be removed for keyboard mode as a whole since it is heavily used in games that need it (SF6 etc.) 

I'm still not convinced that there are modern games that have weird behavior/problems with SOCD off in keyboard mode, but fine. If some people want/need it then it makes sense to leave it in

>2 - Setting the device to keyboard mode and SOCD off should address your concerns, if you are still seeing SOCD rules being implemented then it will be the game / emulator / other settings doing so

Yes, I can entirely disable SOCD globally and then use keyboard mode for rhythm games, as I have been doing. The problem is that SOCD is also off for XInput mode then. I would like it left at a normal/legal setting in XInput (since I use that when playing fighting games on PC) and off for keyboard mode somehow, ideally.
I am not still seeing SOCD applied when it's disabled, I just don't want to get into trouble later for having SOCD off in XInput mode because I needed it off for Keyboard mode. I use them for entirely different things and I don't think I'll ever need SOCD cleaning in Keyboard mode, so being able to separate the setting for them would be fantastic.

TheTrainGoes, 2024-01-12:
Have you considered simply adding a hotkey combo for SOCD cleaning off?  

I've tested this out with the combo of `start` + `home` + `right`= `SOCD Cleaning Off` which will allow you to change to and from it as needed in the various modes.

Soundtoxin, 2024-01-12:
Yes, I have considered it, but I don't actually ever want SOCD Cleaning on in keyboard mode, so ideally I could just set it and forget it instead of having to remember to check/change this setting every time.

TheTrainGoes, 2024-01-12:
Hi @Soundtoxin ,

We have talked internally and have no plans to make SOCD off my default in HID-Keyboard mode.  

This can be easily navigated by setting up a hotkey function to change to SOCD cleaning off as outlined above.

As always, you are welcome to fork the project and make changes as you see fit to your own uses.

closed 2024-01-12 by TheTrainGoes

Soundtoxin, 2024-01-12:
>We have talked internally and have no plans to make SOCD off my default in HID-Keyboard mode.

Right, off by default would conflict with what others want. Some sort of setting in the web configurator (as mentioned upthread a bit) would work fine as well.