OpenStickCommunity/GP2040-CE #785: PS3 port of MvC2 doesn't work on our firmware (as of 0.7.6)

bsstephan, 2024-01-06:
Reported by @TheTrainGoes and tested by me. The Darkstalkers and Jojo's ports work fine, so (so far) seems specific to MvC2 somehow.

I was going to get a PS3 TE1 out of storage at some point and confirm that it works. I was originally thinking I was going to be sifting through a bunch of reports to find differences and try them out, but come to think of it from another conversation, maybe I should just try hacking the VID/PID first and see if that gets us through to the game.

bsstephan, 2024-01-06:
I suppose I should qualify "work", but it's basically any form of button functionality. Turning on LS emulation would let you navigate the top level (e.g. outside of game emulation) main menu, but even so, none of the buttons activated any menus. DPad also didn't do anything. Could still get to the XMB via the Home button, IIRC, and other games worked fine with our firmware before and after, so the controller was fine, just MvC2 didn't care.

TheTrainGoes, 2024-01-06:
To add further to this - has been an issue for a long time.  Although you may want to check v0.4.3 of the original GP2040 to see if it is still an issue there.  On the flip side the home button did not work on GP2040.

bsstephan, 2024-02-15:
Further discussions between us have us collectively curious/interested in if we can get this to work (perhaps justifying a PS3 mode with quirks separate from generic USB HID), so I'm reopening it so it's on my visible list.

PR #840 (2024-02-17): populate the PS3 analog values for buttons
PS3 face buttons are technically analogs like L2/R2. they were declared in the payload but never used, but apparently Marvel vs. Capcom 2 uses the analog values, so use the digital button state to populate the fake "analog" values and get the game working.

this also confirms the order of the axes declared in HIDDescriptors.h, which is nice.

this *ALSO* means MvC2 (and who knows what else) has a predefined notion of what's in the payload bytes despite the interface descriptor, so that's interesting, but whatever

I WANNA TAKE YOU FOR A RIDE

Fixes #7 [... cut]
commit f2d1fae9 (2024-02-17): populate the PS3 analog values for buttons (#840)

PS3 face buttons are technically analogs like L2/R2. they were declared
in the payload but never used, but apparently Marvel vs. Capcom 2 uses
the analog values, so use the digital button state to populate the fake
"analog" values and get the game working.

this also confirms the order of the axes declared in HIDDescriptors.h,
which is nice.

this *ALSO* means MvC2 (and who knows what else) has a predefined notion
of what's in the payload bytes despite the interface descriptor, so
that's interesting, but whatever

I WANNA TAKE YOU FOR A RIDE

 [... cut]
closed 2024-02-17 by arntsonl