OpenStickCommunity/GP2040-CE #807: Add a label to pin mapping profile number

damienleger, 2024-01-19:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

Avoiding to remember which profile corresponds to each game/device. Prevent mistake of selecting the wrong profile. Better UX/QoL for gamer.

### Describe the solution you would like to see

(proto with copy/paste & F12 live edit on v0.7.7 interface)

My idea is to add an optional "label" input on each profile on pin mapping setting (same as the one in macro settings page)

[image]

When set, this label is added in each profile tab on pin mapping setting, and also on the profile selector on the settings page

[image]

[image]

I'm thinking also that the label should be limited to 5 ascii char (good enough imho: sf6, tk8, dbfz, shmup, base, etc.) and added on the display next to the profile number (feature wip: https://github.com/OpenStickCommunity/GP2040-CE/issues/688).

### Describe alternatives you've considered

None.

### Can you submit a pull request?

No

### Are you planning on working on this feature yourself?

No

bsstephan, 2024-02-21:
@TheTrainGoes had an additional idea that we might as well put here: we could also use that label in the OLED.

This would be a fairly simple change, too --- I think you could just add the `name` or `label` as a field to `GpioMappings` and then refer to it in stuff. Seems like a neat idea.

TheTrainGoes, 2024-02-21:
By my count there are 20 character lots on a normal 128x64 display.

Have two fields added in:
[Profile Name] 20 character limit
[Display time length when switching] done in seconds

bsstephan, 2024-03-10:
I think this is my "up next".

PR #1139 (2024-09-25): Add the ability to label a profile and see it in the web UI and OLED display
This adds a 16 character label to the GpioMappings structure, along with managing the label in the web UI (setting it and displaying it in the menu --- thank you again to @Pelsin!) and displaying the label in the OLED display when you switch profiles.

[image]

Resolves #807
closed 2024-09-25 by arntsonl