OpenStickCommunity/GP2040-CE #808: Pin Mapping setting page style uniformization

damienleger, 2024-01-19:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.7
* Name of device: hautepad T16
* Link to where you purchased the device: aliexpress
* Is this a custom build?: no
* Operating System: Win11
* Browser: Firefox 123.0a1

### Expected Behavior

The pin mapping page should be aligned with the overall design of the other pages.

### Current Behavior

In dark mode: the input buttons are blank vs black in other pages.
Input buttons are also bigger than other pages.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. connect to web config ui
2. enforce dark mode if not set by default
3. compare look of pin mapping vs keyboard mapping page

### Screenshots & Files

pin mapping vs keyboard mapping page in dark mode below:

[image]

[image]

Pelsin, 2024-01-19:
Hi!

Thanks for the reporting this issue, totally agree we should make it uniform in style.
The pin page really sticking out like a sore thumb in dark mode.

We are going to look into changing the design and color scheme of the web in the future, but this tiny fix could be done beforehand.

bsstephan, 2024-04-11:
This is improved in v0.7.8, thanks!

closed 2024-04-11 by bsstephan