OpenStickCommunity/GP2040-CE #812: Input history is not always accurate

Cnul971, 2024-01-21:
* Firmware Version: GP2040-CE_0.7.7_Haute42.uf2
* Name of device: Haute42 S16
* Link to where you purchased the device: https://haute42.com/
* Is this a custom build?: no
* Operating System: windows 11
* Browser: Firefox

### Expected Behavior

I used the Macro add to do a quarter circle forward input fast and i expect to see those input display on the LCD

### Current Behavior

The input history drop some input when they follow them each other fast. But the game still recognize the macro well.
The game used is Ultra Street Fighter 4

### Steps to Reproduce

record a quarter circle with each direction last one frame and Post Input Wait Duration last 0 frame

### Screenshots & Files

[image]
[image]
[image]

bsstephan, 2024-02-29:
Unassigning this one from me, because what I found was more wondering aloud than a lead, so someone should take a look at this when they are interested.

TheTrainGoes, 2024-03-28:
@Cnul971 can you please test this again when we release the v0.7.8 RC1 and tell us if its working any better?

Cnul971, 2024-03-29:
@TheTrainGoes ok i will for sure !!! Thanks !!!

Cnul971, 2024-04-02:
@TheTrainGoes tested with RC1, still the same problem (this time with SF6
[image]
[image]
[image]

Cnul971, 2024-04-02:
an other try
[image]
[image]

TheTrainGoes, 2024-06-16:
@Cnul971 have you tested this out in v0.7.9?

TheTrainGoes, 2024-06-21:
Closing - have not gotten word back.  If this is still an issue please file a new issue.

closed 2024-06-21 by TheTrainGoes