OpenStickCommunity/GP2040-CE #816: Addon pin assignment not released after changing via webconfig

MajorArkwolf, 2024-01-22:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.7
* Name of device: OpenCore0
* Link to where you purchased the device: Custom Order
* Is this a custom build?: No
* Operating System: Windows 10
* Browser: Firefox

### Expected Behavior

Please describe the behaviour you are expecting.
When changing the pin assignment via the web config for an addon such as i2c, the previously used pins should become available for use.

### Current Behavior

What is the current behaviour?
When changing the pin assignment via the webconig for an addon, the new pins will be bound, but the old pins will also remain bound even though they are no longer in use.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.
Note: Running these steps will permanently lock your pins out and can not be unlocked via normal means
1. Boot a board into the webconfig
2. Change the D+ pin for the usb passthrough.
3. Save and reboot into the webconfig
4. Go to the Pin Assignment page and notice how the previous and new pins both say they are assigned to an addon.

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

Locutus73, 2024-02-16:
I confirm I had the same issue changing pin assignment in the focus mode add-on with my RP2040 Advanced Breakout Board - USB Passthrough Edition v5.4E and GP2040-CE_0.7.7_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2

arntsonl, 2024-02-16:
I was able to replicate on 0.7.7 using the focus mode pin.

@bsstephan I can take this on or you can grab it.

PR #893 (2024-03-16): unassign the old addon pin after assigning the new one
Fixes #816
commit f5c7fcd5 (2024-03-16): unassign the old addon pin after assigning the new one (#893)

Fixes #816
closed 2024-03-16 by arntsonl