OpenStickCommunity/GP2040-CE #817: Focus Mode blocks macros even when inactive

chunk934, 2024-01-22:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ X] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ X] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [X ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ X] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version:  GP2040-CE_0.7.7_PicoFightingBoard.uf2
* Name of device: Pico Fighting Board based controller
* Link to where you purchased the device: DIY
* Is this a custom build?: YES
* Operating System: WINDOWS
* Browser: OPERA, CHROME, MICROSOFT EDGE

### Expected Behavior

Please describe the behaviour you are expecting.
Focus mode pin in off position allows macros. Focus mode pin in on position locks macros.

### Current Behavior

What is the current behaviour?
Focus mode pin in either position locks macros. Turning off "Lock Macro" enables macros.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Set up macros
2. Set focus mode pin to relevant pin
3. Select "Lock Macro"

### Screenshots & Files
[image]

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

 Tested with focus mode toggle on and off. Tested with different pin for focus mode in on and off. Macros only works when "Lock Macro" is not
selected in focus mode. Tested with 1 firmware older. Flash nuke & Reflashed.

bsstephan, 2024-03-10:
I marked this as a good first issue as it seems like a simple logic oversight.

TheTrainGoes, 2025-01-03:
Duplicate issue, closing

closed 2025-01-03 by TheTrainGoes