OpenStickCommunity/GP2040-CE #818: Left/Right Priority

JJJY5, 2024-01-24:
I wonder if you guys could add a SOCD Cleaning Mode of Left/Right Priority. Since there is Up Priority, why not add Left/Right Priority?

In addition, how could i contribute to the transalation to Chinese?

TheTrainGoes, 2024-01-24:
Hi @JJJY5 ,

What is the use case for Left / Right priority? 

Can you give us an example of a game or situation that would benefit from it?

JJJY5, 2024-01-24:
@TheTrainGoes 

When I was using characters with traditional Shoryoken, such as Luke, Ken, Ryu, Jamie and Cammy, I found that 639(628) is better than 623.If I want to use 639 Shoryoken When crouching(pressing 4 and 2), either I release 2 and press 628p at last win SOCD or I release 2 and 4 and press 628p at neutral SOCD. Obviously last win SOCD is easier for crouching 639 Shoryoken. But if I press 8 and P or K not that simultaneously, Shoryoken will become j.P or j.K.
If there is Left/Right Priority, I dont need to press 8 and P or K at the same time. It will be OK pressing P or K slightly slower than 8.  Dont have to worry about Shoryoken mistakenly become j.P or j.K

Also I noticed that some text of the configuration is not translate into Chinese. I could help.

JJJY5, 2024-01-24:
@TheTrainGoes 
For example, It's easy to use 639 Shoryoken with a normal keyboard(neutral SOCD) when I am not crouching. I could easily press 6 with my left index finger, then press 2 with my left middle finger, lastly press 8 with my left thumb and press P or K with my right index finger. But if I use hitbox with last win SOCD, I must press 2 with my right thumb to make sure P or K is pressed at the same time as 2.

I think add a SOCD should not be very difficult. Some player may like to control 8 only with left thumb, just like me.

TheTrainGoes, 2024-01-24:
This presumes that the character is facing a certain direction correct?  We have no way of knowing the direction a character is facing.

JJJY5, 2024-01-24:
@TheTrainGoes 
I just presume that the character is at 1P(left side) to describe the case. 
I realize that I expressed my request mistakenly. I just want a special last win mode only enable for left and right and disable for up and down. Because the last win for left and right is very useful at most cases.
But last win for up and down(in 99.9% of all cases we press up after down) is useful for only a few characters, like E.Honda or Guile, and sometimes it will become a burden for certain commands, just like the 639 Shoryoken I mentioned.

TheTrainGoes, 2024-01-24:
Hi @JJJY5 ,

There are no plans to introduce SOCD that give Left/Right Priority at this time.

We are going to be closing this as not planned.

closed 2024-01-24 by TheTrainGoes