OpenStickCommunity/GP2040-CE #828: Same hardware, same firmware, different output

FrancescoPenasa, 2024-01-29:
### Prerequisites

- [ x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x ] I am reporting the issue to the correct repository

### Context

This is my third build, while the first two with same hw and same firmware were working great, this third one has buttons all mixed up, for example, insteqd of direction left there is a punch, instead of button L2 there is a direction, even the top button are mixed. I dont know what coule be the problem, I checked the soldering and it seems ok, but i am open to any suggestion

* Firmware Version: 0.7.7
* Name of device: rp2040-zero
* Link to where you purchased the device:
* Is this a custom build?:
* Operating System: window
* Browser: firefox

### Expected Behavior

I expect to have the same output on both devices

### Current Behavior

Different output pressing the samw buttons on two identical devices

### Steps to Reproduce

I dont know, i have done the same steps on bith devices

FrancescoPenasa, 2024-01-29:
I want to add that gamepad tester show the same input mode, the wiring is the same since they are both flatbox v5 and to test it i used the suggested gamepad tester

TheTrainGoes, 2024-01-29:
I am going to guess you flashed a different firmware onto one of them.

Possible that you mixed up the Rev5 firmware with the Waveshare Zero firmware?

If so just put it in BootSel mode, drop the nuke.UF2 on and then put the proper firmware.

FrancescoPenasa, 2024-01-29:
@TheTrainGoes I solved it! thank you, I was using `GP2040-CE_0.7.7_WaveshareZero.uf2` which actually worked fine on another identical system but now I used  `GP2040-CE_0.7.7_FlatboxRev5.uf2` and it works how it was supposed too. I still can't understand why wavesharezero works fine on a system and is mismapped on the other if the flatbox board and the rp2040-zero are the same

closed 2024-01-29 by FrancescoPenasa