OpenStickCommunity/GP2040-CE #829: Allow web configurator port or IP address to be changed

febs, 2024-02-03:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

I have built a two player controller using a 2P enclosure (a "Pandora's box"). I like to only have one USB cable out of it hence I am using an USB hub inside the enclosure, and two GP2040-CE raspberry picos are connected to the hub.
Everything works fine and both controllers are recognized in any system I tried it.

But, if I want to configure both in a PC, if I can't without unplugging and plugging back the USB cable.
It'd  be cool to be able to select a different TCP port for a controller, so when I plug both I can configure one on 192.68.7.1 and the other on 192.168.8.1

### Describe the solution you would like to see

Having a web configurator setting where one can select the IP address of the device.
This could also become handy in case of conflict with home networks on the same default 192.168.7.0/24 subnet

### Describe alternatives you've considered

Unplug and replug the controllers. It's not the end of the world admittedly but having more flexibility would be cool.

### Can you submit a pull request?

Could give that a try

### Are you planning on working on this feature yourself?

Yes. Hopefully wouldn't be too hard to do.

TheTrainGoes, 2024-02-03:
Hi @febs ,

Before you go working on this, there is a shortcut to get into web-config you may be missing.  

While the device is powered on, hold down the `Start` + `P1` + `P2` buttons for 5 seconds.  The device will flip into web-config mode and from web-config you can restart back into controller mode.  

There should be no reason to unplug either unit with this shortcut.

You can read more about access web-config here:
https://gp2040-ce.info/web-configurator/

Regarding the change or possible change of the web-config address, I am a little sheepish to allow the changing of something so core to our setup.

Please let me know if this addresses the concern.

febs, 2024-02-03:
Thanks for your suggestion, which I admittedly didn't think of. It wouldn't be as flexible as my idea, as it would not allow for instance to work on both controllers' configuration at the same time (like, keeping them side to side to compare them and ensure they are alike) but nevertheless could be a workaround.

If you don't feel like changing, then you can happily close the issue.

I eventually could build the binary from source and replace the address by hard-coding it. It would make the thing not as easy to update, but I can live with it ☺

TheTrainGoes, 2024-02-04:
One other thing I will leave you with is that once you set one up the way you like it you can save a backup of the config and apply it to the second unit.

febs, 2024-02-04:
I've Googled for "GP2040-CE export configuration" in vain. Will check that again.

Thanks, will close this

TheTrainGoes, 2024-02-04:
@febs Under configuration -> https://gp2040-ce.info/web-configurator/menu-pages/data-backup-restoration

I will keep this open as its something we may look at adding in the future for advanced config stuff.

TheTrainGoes, 2024-06-21:
We have talked about this further and decided changing the web-config IP is not something we want to support.  

This can always be done with a custom compile if you like but it would be a support nightmare for us if people started changing it and getting locked out or if vendors started making their own IPs set.

closed 2024-06-21 by TheTrainGoes