OpenStickCommunity/GP2040-CE #831: Simple LEDs to signal the toggle of a feature (specifically 4 way joystick)

febs, 2024-02-06:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes.

### Which problem is this feature request solving?

LCD screens are awesome, informative, and very good looking on flatboxes and most modern controllers.
Nevertheless, on old, ignorant, rough boards those wouldn't be really aesthetically fitting in my opinion. Plus, it would just be nice _to have an alternative._
That is, I think it would be cool to be able to just light up a simple, old school LED whenever a feature is enabled or not (for instance, the 4-way joystick mode, or if profile 1 is selected, whatever).

### Describe the solution you would like to see

Being able to bind the lighting/turning off or on of a led to a pin and a feature, such as the 4 way mode.

By the way, specifically 4 way mode isn't featured _at all_ in the I2C display elements.

### Describe alternatives you've considered

I don't know really. I found myself having an harder time than expected to realize whether my joystick is in one mode or another.
While browsing a retrogaming console interface there's no way to tell. I have to open a game to actually test that.

Plus it would be super cool to have a red LED on when the stick is 4 way :-D

### Can you submit a pull request?

Should be able to.

### Are you planning on working on this feature yourself?

Can try.

febs, 2024-02-06:
Speaking of which, I might have found an involuntarily cut of the Turbo LED documentation:
[image]

bsstephan, 2024-02-21:
I moved the docs issue into a new defect.

TheTrainGoes, 2025-07-24:
We are going to close this as not planned

closed 2025-07-24 by TheTrainGoes