OpenStickCommunity/GP2040-CE #843: turbo for adc problem

kumanecoyaro, 2024-02-17:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version:7.7
* Name of device:official pico
* Link to where you purchased the device:
* Is this a custom build?:no
* Operating System:win10
* Browser:vivaldi

### Expected Behavior

turbo shot for setting

### Current Behavior

when turbo adc mode is on by setting with web config but that uses add big latencey.system is berry slow.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Step 1
2. Step 2
3. ETC...

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

TheTrainGoes, 2024-02-21:
Not enough information in this.

@kumanecoyaro please do a better job filling out the issue or this will be closed.

bsstephan, 2024-02-21:
Marking this invalid until/unless we get more info. If we don't hear back soonish one of us will indeed close this.

kumanecoyaro, 2024-02-22:
1.I was wire V_ref,gnd,num26pin with volume register(10k ohrm)
2.Setting at web-comfiglator as turbo that adc from -1 to 26
3.I plugged to USB to PC...etc
after effect gp2040ce running berry slow and big latency

FeralAI, 2024-02-25:
To clarify the issue here, when the turbo dial is enabled there are analog reads happening on each call to process() of the Turbo addon. This will be address in https://github.com/OpenStickCommunity/GP2040-CE/pull/863.

bsstephan, 2024-03-10:
This has been addressed and will be in the 0.7.8 release; closing.

closed 2024-03-10 by bsstephan