OpenStickCommunity/GP2040-CE #844: Adding Support for Neo Geo Arcade Stick Pro emulation so 6 buttons can be used instead of 4 for 2 players

pelana123, 2024-02-18:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

Yes 

### Which problem is this feature request solving?
Im able to use neo geo mini controller mode but its limited too 4 buttons and 6 are achivable if another Arcadestick pro is connected so id like to be able to emulate an Arcadestick pro controller in neo geo/pc mode 

### Describe the solution you would like to see
I think that adding ASP support as an option will help many people improve their experience with the console as it is a great option too play games but lacks a cheap way to add a 2nd player with full button support 

### Describe alternatives you've considered

they say mayflash has it available in an update but im not sure about it 

### Can you submit a pull request?

/No. I dont understand what this means 

### Are you planning on working on this feature yourself?
I dont know how to do it myself or i would do it and share it but  i would gladly help gather any info from the device thats needed and would love to test as i have many (5) devices using gp2040-ce i could do testing with

TheTrainGoes, 2024-02-18:
Hi @pelana123,

I am having a bit of trouble following what the issue is here.

If you have two devices connected to an ASP both running in NeoGeo Mini mode they should each have access to four buttons, and that should be all that is needed for the games which are included on the ASP.

Are you running a modified ASP that have games which need more than 4 buttons?

pelana123, 2024-02-18:
Yes, im running a modded ASP with games that need more than 4 buttons

TheTrainGoes, 2024-02-18:
Hi @pelana123 ,

We will consider that use case to be largely out of scope for the implementation of the mode since the mode is working as intended for a stock ASP.   

Having said that we will see if there are any minor changes that can be made to add that functionality without any detriment to the original functioning.

PR #845 (2024-02-21): Added button masks for Neo Geo ASP extended buttons
This should allow the side panel Options button and 4 extra action buttons to be included where used.

Assume the following mapping, as the remaining 4 buttons are unlabeled (using L1/L2/R1/R2 naming).
```
A C R1 L1
B D R2 L2

B3=A
B1=B
B4=C
B2=D
S1=Select
S2=Start
A1=Options
L1=L1
L2=L2
R1=R1
R2=R2
```
This should help to resolve #844 .
closed 2024-02-21 by arntsonl