OpenStickCommunity/GP2040-CE #846: Enabling Turbo Dial cause instability problems

fostrike, 2024-02-19:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: v0.7.7
* Name of device: RP2040-Zero
* Link to where you purchased the device:
* Is this a custom build?: yes
* Operating System: Win11
* Browser: Chrome

### Expected Behavior

10K potentiometer connected to analog I/O pin26 to dial the turbo speed

### Current Behavior

So i tested the use of a Turbo button, a Turbo LED and a Turbo Dial function back in September 2023 on firmware version V.7.1.0 with a simple bread board.
Everything was fine and working as expected so i made a custom PCB.
Now i just assembled the PCB and flashed the new WaveshareZero firmware v7.7.0.
The Turbo functions are still working as expected but Windows very often fail to properly recognize the device, giving the "unknown device" error.
This problem goes away as soon i disable the Turbo Dial on pin26 in the web-config.
On top of that, if i plug the USB while the potentiometer is set to full turbo speed the configuration made in the web interface are wiped to the default settings.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Just enabling the Turbo Dial on an analog I/O pin make the device pretty unstable and Win11 and Win10 often fail to recognize the device

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

bsstephan, 2024-02-21:
@fostrike this sounds legitimate based on your test with v0.7.1, but for completeness's sake, have you tried v0.7.7 on your breadboard setup?

fostrike, 2024-02-21:
No but i did isolate the 3,3V output path that in the PCB goes to the Turbo LED trough a resistor and the 10K pot for the analog pin 26.
So only the potentiometer and few buttons are connected, the OLED pins are wired but unconnected.
If i measure with a multimeter between the RP2040Zero 3.,3v out pin and pin 26 i got the full range of the pot, from 0 to 10K ohms

FeralAI, 2024-04-11:
Hello @fostrike! Can you test out the latest release [v0.7.8](https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.8)? There were several updates to turbo functionality, and specifically there was a bug that was fixed when using a turbo dial that may have caused the behavior you reported.

fostrike, 2024-04-12:
Hello Feral, first of all thanks for the fixes.
The instability problem seems resolved, i can leave the dial knob at any level and the controller is always properly recognize by Windows.
The only thing is that by default when i plug the device the Turbo LED is always ON, then when I assign the Turbo function to a button the LED goes OFF and work as usual, blinking according to the Dial adjustment.
Is this the expected behaviour?
In my wiring i got the 3.3V connectet to the LED positive and the negative goes to the RP2040 pin, so i suppost at startup this pin goes to 0?

FeralAI, 2024-04-13:
> Hello Feral, first of all thanks for the fixes. The instability problem seems resolved, i can leave the dial knob at any level and the controller is always properly recognize by Windows. The only [...]

Thanks for testing this out, and good to know it fixed that locking up issue. Regarding the LED, the logic in the turbo code is expecting the positive to connect to GPIO. This is because the breakout boards for the project use a generic 2-pin JST that connect a GPIO and ground. My test setup is similar to yours with LED- connected to GPIO. In the next version I'll add a toggle to the turbo addon that will allow inverting the turbo LED logic to correct this.

closed 2026-10-22 by TheTrainGoes