OpenStickCommunity/GP2040-CE #850: Buttons keep double-pressing.

Parindraa, 2024-02-22:
### Prerequisites

Please check the following before posting an issue / bug report.

- [X ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [X ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ X] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [X ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: GP2040-CE_0.7.7_PicoW.uf2
* Name of device: Pico W
* Link to where you purchased the device: [https://www.aliexpress.us/item/3256806007268104.html?spm=a2g0o.productlist.main.3.32a01a46mfMz2n&algo_pvid=4d303535-25e5-43ad-bb71-44ad6f722e34&algo_exp_id=4d303535-25e5-43ad-bb71-44ad6f722e34-1&pdp_npi=4%40dis%21USD%215.70%212.69%21%21%215.70%212.69%21%402101fb1617085695173341731eaf78%2112000036213854354%21sea%21US%210%21AB&curPageLogUid=pXDyxueN89Ft&utparam-url=scene%3Asearch%7Cquery_from%3A](url)
* Is this a custom build?: No
* Operating System: Retropie
* Browser: Chromium??

### Expected Behavior
Momentary push buttons are not supposed to double press and everything functions normally.

### Current Behavior
Push buttons double press whenever I press a button. Quickly jabbing at the button would make it press only once. If I were to hold the push button, it functions as expected, double pressing at a normal speed.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Press a button like you would normally.
2. Wait and see the results!
3. Suprise!

TheTrainGoes, 2024-02-22:
Hi @Parindraa ,

Is it like this only in menu or also in game?

Have you tried increasing the debounce setting?

bsstephan, 2024-02-24:
@Parindraa - if possible, can you try the branch in this PR?: https://github.com/OpenStickCommunity/GP2040-CE/pull/857

bsstephan, 2024-03-09:
Closing as we haven't heard back, but this is believed to be fixed in 0.7.8 as the above PR has merged.

closed 2024-03-09 by bsstephan