OpenStickCommunity/GP2040-CE #861: I accidentally disabled the start button in the web configurator and now I can’t access the settings

NiRi88, 2024-02-24:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version:
* Name of device:
* Link to where you purchased the device:
* Is this a custom build?:
* Operating System:
* Browser:

### Expected Behavior

Please describe the behaviour you are expecting.

### Current Behavior

What is the current behaviour?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Step 1
2. Step 2
3. ETC...

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

bsstephan, 2024-02-24:
You will have to nuke your board and reinstall the firmware and reconfigure it, or do more complicated edits of the configuration via direct flash access (which is possible but complicated).

closed 2024-02-24 by bsstephan

bsstephan, 2024-02-25:
@NiRi88 I was reminded that there is a UF2 that forces you into webconfig mode: https://gp2040-ce.info/assets/files/force_webconfig-347a1a8575da1ae314cf9ceb7df793bc.uf2 --- copy it onto your device in BOOTSEL mode (do not nuke it).

Either way, this is probably more appropriate for the discussions rather than an issue, so I'm keeping this closed.