OpenStickCommunity/GP2040-CE #867: Saving Protobuf breaks Core0 functionality

arntsonl, 2024-02-27:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: dev
* Name of device: n/a
* Link to where you purchased the device:
* Is this a custom build?:
* Operating System:
* Browser:

### Expected Behavior

Protobuf should be able to save via the Save() system without breaking any functionality on the GP2040 project.

### Current Behavior

Save() causes everything to lock up including the tud_task and tuh_task calls because of how complex protobuf is. This should be moved over to Core1 if possible to leave Core0's usb processing abilities.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Turn on PS passthrough and plug in a dongle
2. Change led animations
3. Observe the ps passthrough device reconnects
4. Change turbo speed
5. Observe the ps passthrough device reconnects

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

arntsonl, 2024-02-27:
I'll take point on this one since I've been doing a lot of core revamping.

bsstephan, 2024-02-29:
Is it because protobuf is complicated, or is it because FlashPROM locks the cores when writing?

bsstephan, 2024-02-29:
@arntsonl --- https://github.com/OpenStickCommunity/GP2040-CE/blob/main/lib/FlashPROM/src/FlashPROM.cpp#L16 is what I mean, by the way. I would have to look at nanopb but I would be surprised if the serialization was a major problem, since the config is already in a struct at runtime and just has to be packed in order to be written.

arntsonl, 2024-02-29:
Oooooh I'm not sure! That will have to be part of the investigation

TheTrainGoes, 2025-05-22:
Work was completed on this.

closed 2025-05-22 by TheTrainGoes