OpenStickCommunity/GP2040-CE #870: Strange Pin Response After Update

ghoticat, 2024-02-29:
### Prerequisites

Please check the following before posting an issue / bug report.

- [X ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ X] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [X ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [X ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: v.0.7.7
* Name of device: haute 42
* Link to where you purchased the device: https://www.amazon.com/dp/B0CN6M44F4?ref_=cm_sw_r_apan_dp_2G4FK0H77Z7EANP66ZFG&language=en-US&th=1
* Is this a custom build?: not sure
* Operating System: Windows 11
* Browser: Google Chrome

### Expected Behavior

I expect the buttons I press to send the signals intended to whatever device it's currently plugged into,

### Current Behavior

I wasn't having any issues initially. However, since I updated to v.0.7.7 I seem to be experiencing a strange discrepancy with how the pins are responding. When I use the "pin mapping" feature in the web configurator, the pins respond as expected when pressed. However, after I save & close out of the configurator, the controller only gives me pin 26 if I press pin 21, and it only registers pin 19 if I press pins 26 & 19 simultaneously. For reference, I also have pin 18 mapped to the same controller button as pin 26 (as this is the default settings), and all of what I mentioned previously is still true when pressing pin 18 instead of 26.

### Steps to Reproduce

I haven't changed any settings since I updated to v0.7.7. I even used the "reset settings" feature 

### Screenshots & Files

[image]

[image]

TheTrainGoes, 2024-02-29:
Hi @ghoticat ,

Please nuke and reload the v0.7.7 firmware then report back if the issue persists.  

The nuke file can be found here:
https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.7/flash_nuke.uf2

ghoticat, 2024-02-29:
Thanks for your quick response! I tried this just now, and it still seems to be having the same issue. I even tried redownloading the flash nuke & haute42 0.7.7 update just incase I somehow managed to download the wrong files.

ghoticat, 2024-03-01:
Actually, I apologize. Seems like after further testing, everything is working as normal. Boy, do I feel dumb! Thanks for your help!

closed 2024-03-01 by TheTrainGoes