OpenStickCommunity/GP2040-CE #871: Pin Viewer is unable to show GPIO 23, 24, 25, 29

CapnHawke, 2024-02-29:
### Prerequisites

Please check the following before posting an issue / bug report.

- [X] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [X] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [X] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [X] I am reporting the issue to the correct repository

### Context 

* Firmware Version: [GP2040-CE_0.7.7_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2](https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.7/GP2040-CE_0.7.7_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2)
* Name of device: RP2040 Advanced Breakout Board Version 5.5e Passthrough Edition
* Link to where you purchased the device: JLCPCB
* Is this a custom build?: No
* Operating System: Windows 11
* Browser: Google Chrome

### Expected Behavior

When using the Pin Viewer function at http://192.168.7.1/pin-mapping I expect the pin viewer to confirm whenever a GPIO pin connects to ground/a button is pressed. 

### Current Behavior

When using the Pin Viewer function, the pin viewer displays the pin corresponding to the pressed button, except for GPIO 23, 24, 25, and 29. For each of those four pins, the pin viewer times out waiting for an input and it appears that any button presses occurring on those pins simply do not register.

### Steps to Reproduce

1. Move the USB slide switch from "USB" to "Options"
2. Connect wires to a known working button, wire the button to G and 23.
3. Navigate to the Web Configuration tool, pin mapper, then press the pin viewer button. 
4. When prompted for a button press, press the button connected to G and 23. The pin viewer will not respond and will eventually time out.
5. Assign an input to Pin 23, e.g., "up", save, and reboot in controller mode.
6. Navigate to a gamepad tester.
7. Press the button and watch as the pin delivers the desired input notwithstanding that the pin viewer was unable to detect it. 

Repeat the steps above for pins 23, 24, 25 (with the LED slide switch configured correctly), and 29. 

### Screenshots & Files

Video of the pin mapper failing to register on Pin 23: https://cdn.discordapp.com/attachments/1049370387563163731/1212664412083716107/Pin_viewer_failure.mp4
Video of the button press succeeding notwithstanding that the pin is invisible to the Pin Viewer:
https://cdn.discordapp.com/attachments/1049370387563163731/1212664506820329492/Button_press_success.mp4

bsstephan, 2024-03-09:
@CapnHawke are all of those pins assigned to addons elsewhere in the config?

CapnHawke, 2024-03-09:
> @CapnHawke are all of those pins assigned to addons elsewhere in the config?

At the time I originally found this issue, I had pin 23 and 24 used by the USB Host.

As to pin 25, I am unsure how to best answer the question. Pin 25 looks to be mapped to the On-Board LED in the add-on configuration menu.  I have not found a way to select any other pin, or to release the pin assigned to that function. Unlike other add-ons, there is not a field available to enter a pin number. I first experienced this issue nine days ago, and I do not remember whether I was experiencing it at a time when the On-Board LED addon was enabled or disabled. So I tested again today.

Pin 29 was not and never has been mapped to an add-on while experiencing this issue.

Today's test: 

1. Flash Nuke
2. Install new instance of GP2040-CE_0.7.7_RP2040AdvancedBreakoutBoardUSBPassthrough.uf2
3. Navigate to web config
4. Test pin mapper, confirm that 23, 24, 25, 29 are not detected.
5. Navigate to USB Host configuration field, assign pin to -1.
6. Disable USB Host.
7. Test pin mapper, confirm that 23, 24 continue to be undetected.
8. Verify that the On Board LED in add-ons is disabled.
9. Test pin mapper, confirm that 25 is undetected.
10. Test pin 29, confirm that it is undetected by the pin mapper.
11. Search through add-ons to confirm that pin 29 is not assigned to anything. (as expected in a fresh instance of the firmware)
12. Just for fun and information searching, map pin 29 to turbo.
13. Test pin 29. It is still undetected by the pin mapper.

bsstephan, 2024-03-09:
If you reboot the webconfig mode after step 6, does that make 23 and 24 work in the pin mapper?

CapnHawke, 2024-03-09:
Just ran through the steps from the beginning, saved, and rebooted into web config after step 6. Pin viewer still does not show these pins.

To be clear, the pins do function.

bsstephan, 2024-03-10:
Odd, but we'll continue to investigate.

CapnHawke, 2024-03-10:
I have an update on this, isolating the issue. 

1. Flash Nuke
2. Install fresh instance of 0.7.7 firmware
3. Navigate to Web Config.
4. Click Pin Viewer, and test pins 23, 24, 25, and 29. Observe that they are not detected, the Pin Viewer times out.
5. Disable addons using pins 23, 24, and 25.
6. Map inputs to pins 23, 24, 25, and 29. It doesn't seem to matter what. Could be DDI directions, could be B1-B4.
7. Save.
8. Restart Board in web config mode.
9. Test pin viewer, and observe that pins 23, 24, 25, and 29 **are** detected by the pin viewer.
10. Remove the mapped inputs on 23, 24, 25, and 29, assigning them to "none."
11. Save.
12. Restart board in web config mode.
13. Test pin viewer, and observe that pins 23, 24, 25, and 29 **are once again not detected** by the pin viewer.

It appears that the pin viewer is unable to detect pins 23, 24, 25, and 29, when they are not mapped to a function in the pin viewer. However, the pin viewer is able to detect other unmapped pins. For example, pins 26, 27, 2, and 3 can all be detected even when they have no assignment to any function.

looper1234, 2024-04-19:
[image]
those pins are not useable it seems 23,24,25

CapnHawke, 2024-04-19:
> [image]
> 
> those pins are not useable it seems 23,24,25
> 
> perhaps its better to grey them out in the mapping tool

Thank you for your interest but this is not the hardware being discussed. The PCB in question is designed differently and can access those pins.

looper1234, 2024-04-19:
> > ![5$JC59H_CVZV5Q{3SQ0KVWF](https://private-user-images.githubusercontent.com/10213191/323935782-68f0ece3-2d8d-4355-b3d0-0318b50d9c0d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1MzUxMDUsIm5iZiI6MTcxMzUzNDgwNSwicGF0aCI6Ii8xMDIxMzE5MS8zMjM5MzU3ODItNjhmMGVjZTMtMmQ4ZC00MzU1LWIzZDAtMDMxOGI1MGQ5YzBkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE5VDEzNTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExZWM2Yjk2YmE0OWMxMDMwNDQwNTQxYTkwYzY1NDM0ZGQ1MzljYjViNzIxMWNlNmY5YjVmNDI2YTBjY2ExNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xrmi5arV4N7PdbkiV4z11cm4vNyd0_ueMxjWP-Q-I0Y)
> > those pins are not useable it seems 23,24,25
> > perhaps its better to grey them out in the mapping tool
> 
> Thank you for your interest but this is not the hardware being discussed. The PCB in question is designed differently and can access those pins.

I wanted to add more buttons to my hitbox, thinking 23,24,25 are available, but they are not.

after looking into it for a while led me to this:
https://github.com/pinout-xyz/pico.pinout.xyz/issues/13

from the schematics posted there, GPIO23,24,25,29 all have hardware purpose and are thus wired differently for their purpose,
might not be related to your problem, but seems like too big of a coincidence

the current firmware shows these pins as customisable, but thats most likely not true as you won't be able to connect a button to them

CapnHawke, 2024-04-19:
> I wanted to add more buttons to my hitbox, thinking 23,24,25 are available, but they are not.
> 
> 
> 
> after looking into it for a while led me to this:
> 
> https://github.com/pinout-xyz/pico.pinout.xyz/issues/13
> 
> 
> 
> from the schematics posted there, GPIO23,24,25,29 all have hardware purpose and are thus wired differently for their purpose,
> 
> might not be related to your problem, but seems like too big of a coincidence
> 
> 
> 
> the current firmware shows pin 25 as customisable, rest as addons, but thats most likely not true as you won't be able to connect a button to 25

You may simply need a different PCB. The schematic for the PCB being discussed in this issue is pictured here.

[image]

The buttons which have been mapped to these pins function just fine. They also show up in the pin viewer after they have been assigned to an input. The sole issue is that while these pins are not assigned, they are not visible to the pin viewer while other pins are.

looper1234, 2024-04-19:
> > I wanted to add more buttons to my hitbox, thinking 23,24,25 are available, but they are not.
> > after looking into it for a while led me to this:
> > [pinout-xyz/pico.pinout.xyz#13](https://github.com/pinout-xyz/pico.pinout.xyz/issues/13)
> > from the schematics posted there, GPIO23,24,25,29 all have hardware purpose and are thus wired differently for their purpose,
> > might not be related to your problem, but seems like too big of a coincidence
> > the current firmware shows pin 25 as customisable, rest as addons, but thats most likely not true as you won't be able to connect a button to 25
> 
> You may simply need a different PCB. The schematic for the PCB being discussed in this issue is pictured here.
> 
> ![image](https://private-user-images.githubusercontent.com/140472147/324036490-015b7092-a2bd-4bb9-bc82-9f0c70a02448.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1NDY3MTAsIm5iZiI6MTcxMzU0NjQxMCwicGF0aCI6Ii8xNDA0NzIxNDcvMzI0MDM2NDkwLTAxNWI3MDkyLWEyYmQtNGJiOS1iYzgyLTlmMGM3MGEwMjQ0OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxOVQxNzA2NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wOTY4ODZjZjcxNjI1NTZmOGJmZGY5ZmVhMjM2YzQxZDAzZjM0YzdmY2FlZjQ1NzUwZTAxOTk1NjA2NjZkZjNmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.OKMPoVdVsmrwzLrwqNomf2IdpWpkV9-fGhb_R1sQhe4)
> 
> The buttons which have been mapped to these pins function just fine. They also show up in the pin viewer after they have been assigned to an input. The sole issue is that while these pins are not assigned, they are not visible to the pin viewer while other pins are.

I am only using the standard pico board
[image]
from its documentation, the pins you are having problems with are listed as internal use

i can't find the wiring diagam for the breakout board, thus cannot confirm if they share these uses

looper1234, 2024-04-19:
[image]
found it, 25 is the LED the rest are OPTIONS 

don't know what they are for tho

CapnHawke, 2024-04-19:
> [image]
> 
> found it, 25 is the LED the rest are OPTIONS 
> 
> 
> 
> don't know what they are for tho

GPIO 25 can be used for other things too. I suggest we leave the "issue" alone and you can join the GP2040-CE discord if you are interested in learning more about what your basic pico can do, and what the RP2040 Advanced Breakout Board (and its revisions) can do. The capabilities of these boards are maybe a discussion better had somewhere else. Same with your controller. 

https://discord.gg/openstickcommunity-1049366310389289001

arntsonl, 2025-01-25:
Here's the issue:

[image]