OpenStickCommunity/GP2040-CE #873: Pin debouncer on xbox series x - Guide menu dont work

strusic, 2024-03-01:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: from [HERE](https://github.com/OpenStickCommunity/GP2040-CE/actions/runs/8025923440)
* Name of device: Flatbox REV5 - custom build with oled, rgb and usb passthrough
* Link to where you purchased the device: done it myself
* Is this a custom build?: as of board? yes
* Operating System: win10, xbox series x
* Browser: edge, opera

### Expected Behavior

After uploading fix, it should fix things. It does but not as it should

### Current Behavior

I was testing this firmware with debouncing fix on xbox series x in xbox one mode. All buttons working well, double press dissapear but X button (aka Home or Guide - A1). It is pressed as shown on oled screen, but there are no reaction (Home menu is not appearing), and after few clicks I have Power Menu - the one when you hold Home menu for 2-3 secconds.

### Steps to Reproduce

Just plug your device with Magic-X inserted in xbox one mode with this firmware, and try to press A1 button

### Screenshots & Files

None

arntsonl, 2024-03-01:
Thank you! Yeah I'm guessing its something on the virtual keycode stuff I had on the old code. Good news is I know where it is in code, just need to finish up my other big backlog of code.

arntsonl, 2024-04-02:
This is fixed for sure in latest. Will be in RC release

closed 2024-04-02 by arntsonl

strusic, 2024-04-02:
Yes it is. I build main few days ago to check that :)