OpenStickCommunity/GP2040-CE #876: Fn + Trigger Button = Output Button, but cannot output long press

jack2game, 2024-03-02:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: Compiled from latest [main](https://github.com/OpenStickCommunity/GP2040-CE/commit/53099a722f1ed21f222825b75e32c56c218239a7)
* Name of device: -
* Link to where you purchased the device: -
* Is this a custom build?: -
* Operating System: Windows 11
* Browser: Chrome

### Expected Behavior

When using Fn as a shift key, such example, Fn + A = RT, it should be able to output long-press based on hold long Button A is pressed.

### Current Behavior

Output only last for a split second

bsstephan, 2024-03-03:
There are hotkeys that should not be held (like toggling 4-way) so they all only trigger once right now. This will have to be addressed as a feature of some sort.

jack2game, 2024-03-03:
I’m sorry but I believe this is actually a serious bug rather than a potential enhancement. It was working fine in the previous version of the firmware. 

If one can trigger the output button by one frame (as opposed to how long the physical button was being held), half of the Fn functions are almost useless, including all those that output another button.

bsstephan, 2024-03-03:
You're right; the code I wrote to not repeat the action was before (most of) the buttons were added, but I apparently thought ahead and only applied it to toggles and other things that were saving, to avoid unnecessary save calls. However it looks like this regressed in 43358ed8f and got applied to all hotkeys.

CC @arntsonl

TheTrainGoes, 2024-03-03:
Hi @jack2game,

I would not consider this a serious bug, rather an unintended change based on some other changes that were made.  I could just as easily argue that FN button activations were never designed to be held.  

Feel free to submit a PR with the fix on it if you like, else we will get to it when we have some time.

PR #906 (2024-03-22): Allow some hotkeys to be held again
Some hotkeys are just chords of a sort for normal button outputs, and should be allowed to be held. Some refactoring accidentally undid that and had all hotkeys be checked against the previously run hotkey, such that they all only ran once until the hotkey was let go. This restores the old functionality and also guards some additional hotkeys that don't need to be held and should only run once because they are toggles and/or they save to flash.

Fixes #876
commit 540b9f4d (2024-03-22): Allow some hotkeys to be held again (#906)

* allow some (most, actually) hotkeys to be repeatable again

Fixes #876

* guard more option-saving hotkeys behind action != lastAction

this is just to reduce unnecessary saves if for some reason these
hotkeys are held for a long time
closed 2024-03-22 by arntsonl