OpenStickCommunity/GP2040-CE #884: Button L2 adn R2 trigger button

hermogenes12, 2024-03-08:
I'm using the rp2040-ce in a project for a portable PS3 because it was the best solution I found for control since the Dualshock 3 itself had a lot of problems. But I noticed that in some games the L2 and R2 buttons don't work with the triggers button but only as normal buttons. That said, I would like some help implementing these trigger buttons to finalize my project and present my work to the 3D printing community. Noting that in PS4 mode the buttons normally function as trigger buttons.

Eu estou usando o rp2040-ce em um projeto de um PS3 portatil por que foi a melhor solução que encontrei para controle visto que o proprio Dualshock 3 apresentava muito problema.Mas notei que em alguns jogos o botão L2 e R2 não funcionam com botao gatilhos mas apenas como botoes normais.Dito isto gostaria de uma ajuda para a implementação desses botoes gatilhos para finalizar meu projeto e apresentar meu trabalho para a comunidade de impressão 3D.Notando que no modo ps4 os botões funcionam normalmente como botao gatilhos.

TheTrainGoes, 2024-03-08:
Hi @hermogenes12 ,

Can you please give us more information about the issue.

What version of the firmware are you using?
What game are you seeing this issue in?
Is your PS3 modded?

hermogenes12, 2024-03-08:
Firmware  = GP2040-CE_0.7.5
Games  = God Of War Ascencion,Castlevania-Lords-of-Shadow
Yes Ps3 Modded?

bsstephan, 2024-03-08:
Just sharing this here that the PS4 mode exposes the L2/R2 as axes and the PS3 mode does not, that may be the discrepancy here.

bsstephan, 2024-03-09:
Based on my hunch I'll try to find a game that has L2/R2 behavior that isn't a button press (meaning they plausibly use the analog axes (and presumably also look for them in the descriptor)), but I don't have the ones listed, so further information would be nice.

TheTrainGoes, 2025-01-03:
@hermogenes12 is this resolved in v0.7.10?