OpenStickCommunity/GP2040-CE #885: Bug - Mapping Function key in Keyboard Mapping resets to None after save and reboot

jake-najera, 2024-03-08:
### Expected Behavior

- Would expect that changing the Function key from None to "Q", would result in the device outputting "Q" on key press and the Web Config to update noting that change after save / reboot.

### Current Behavior

- Binding the function key in the Keyboard Mapping menu and pressing save clears any value selected in its' dropdown menu.

### Context

- Device: Custom board with a Pi Pico.
- Firmware and Distribution version: GP2040-CE_0.7.7_Pico.uf2
- OS: Windows 10
- Browser: Chrome

- Support thread in Discord with Images: https://discord.com/channels/1049366310389289001/1215741350868160692

### Steps to Reproduce

1. Enter web-config
2. Configuration > Keyboard Mapping
3. Change Function from None to non-conflicting key
4. Save
5. Reboot to Controller
6. Test in Keyboard Tester

The support thread linked above has screenshots of the Console after save.

bsstephan, 2024-03-09:
Sounds like a duplicate of #836. Please actually test the Function key functions, despite what the web UI says, and report back/close.

artemis-u, 2024-03-09:
Mapped Function key to "Q", saved and rebooted, and then used [Keyboard Tester](https://www.keyboardtester.com/tester.html) to detect keyboard inputs. Still gave no output when pressing the button.

bsstephan, 2024-03-09:
Oh, Function specifically, that isn't intended to be output anyway. It serves as a shift key or service mode rocker switch for the firmware hotkeys. It probably shouldn't be mappable as an output key.

bsstephan, 2024-04-11:
Fn has been made unmappable in v0.7.8, thanks!