OpenStickCommunity/GP2040-CE #895: PWM Keyboard Output?

DeadlyDad, 2024-03-16:
(I read through the docs but didn't find a 'PWM Keyboard Output' function, so I'm posting this.  I may have missed it, though.  :shrug: )

More than twelve years ago, I wanted to use my MS FF steering wheel with games that didn't support it, so I installed an app called [GlovePIE](https://web.archive.org/web/20150204054412/http://glovepie.org/) and hacked together a simple script that translated analog control positions (steering/accel/brakes/etc.) into PWM keyboard signals.  

It was [pretty crude](https://web.archive.org/web/20150205011622/http://www.glovepie.org/w/index.php?title=Some_Examples#Wheel2WASD) but got the job done so well there was absolutely **NO** detectable difference between playing my racing games in native (non-FF) wheel mode and using the script with the game set up for keyboard control.  (It was actually kind of eerie just how well it worked.  It was also awesomely hilarious to be using a full-on force-feedback steering wheel with (emulated) SNES and Flash racing games. :rofl: )

I **STRONGLY** suggest adding a similar feature to this project, as it would allow full analog (driver-free!) control with ***any*** hardware that will support a USB keyboard.

Cheers!

bsstephan, 2024-03-17:
What hardware/game supports PWM keyboards but doesn't support USB HID controllers?

DeadlyDad, 2024-03-18:
1. Not 'PWM keyboards'; just 'keyboards'.  I'm not aware of any keyboards using linear key switches that use PWM, though that is both an obvious and useful idea.
2. Emulated consoles/browser games/etc. tend to expect at least partial digital controls, and PWM keyboard control allows for things like using an analog joystick to--for example--set the walking pace of characters even in (emulated) NES games or use a steering wheel to control racing cars in a Flash game.  You could also use it for turbo fire by setting an analog trigger to PWM control a digital input by using the analog value to set the length of the cycle instead of the pulse.

Cheers!

DeadlyDad, 2024-03-26:
> Hi @DeadlyDad ,
> 
> Can you expand on this a bit? Are you looking to use a specific device and want to use it through GP2040-CE or are you looking to build using the features you have listed? We support most of those things already, just not through keyboard passthrough.

1. I have a [Logitech Attack 3 joystick](https://www.bhphotovideo.com/c/product/597774-REG/Logitech_963291_0403_Attack_3_Joystick.html) that I would love to use with my left hand for better movement control and my mouse with my right, but 90% of games make you choose either keyboard and mouse **OR** controller.  (For the best experience when playing shooters, you need both analog directional **and** positional controls, and neither keyboard & mouse nor controller combines the two.)
2. When I opened my joystick up, I found that there is _plenty_ of space to swap in an RP2040/etc. for the original controller board.  (...and swap in Hall Effect pots, too!!!)
3. If GP2040-CE offered PWM keyboard output, I could just set every game to k&m and _still_ have analog movement control where the joystick X/Y was used for PWM-control of WASD.
4. I'm not just looking for _my_ problem, but a common problem with most games, for which this feature would be a **godsend**.

Does that help explain things?  (I honestly can't tell.  I've been under so much stress lately, I don't know if I'm communicating properly or not.)

TheTrainGoes, 2024-03-26:
Hi @DeadlyDad 

Have you checked out the HID remapper project from Jfedor?  It may suit your needs better:
https://github.com/jfedor2/hid-remapper
https://github.com/jfedor2/hid-remapper/tree/master/custom-boards

TheTrainGoes, 2024-04-15:
Never heard back on this, going to close as not planned.