OpenStickCommunity/GP2040-CE #896: GP2040-CE 0.7.7 Oled screen configuration problem.

TNiks, 2024-03-16:
Device :Homemade flatbox  WAVESHARE RP2040 Zero and 0.96" oled screen 128X64 I2C IIC

Firmware : GP2040-CE_0.7.5_WaveshareZero
The problem that I am having is that when I try to upgrade to 0.7.7 with nuke install , the screen doesn't work anymore.
According to other users experience when I posted the issue to discord support channel https://discordapp.com/channels/1049366310389289001/1212816199742656522 from firmware version 0.7.6 and onwards the i2c1 isn't working for my device and probably and some others .There isn't any issue with i2c0 when assigning pins 00-01 as descirbed at discord support channel but is not convenient to use them with the wave share board  .
Other users at the support channel were able to reproduce the problem described at the support channel.

bsstephan, 2024-03-17:
Pending investigations, though reproduction has been spotty at the moment.

TNiks, 2024-04-15:
> @TNiks <https://github.com/TNiks> have you tested this on the new v0.7.8?

Yes I have tested 0.7.8 and works flawlessly.Good job!!!

closed 2024-04-15 by bsstephan