OpenStickCommunity/GP2040-CE #901: Github White Spaces Turned Off?

arntsonl, 2024-03-19:
Developer Issue:

Github white spaces was turned off it looks like. This is a developer note, we need to make sure white spaces is correct so we do not have check-ins with tabs, 2 white spaces, 4 white spaces, etc.

Pelsin, 2024-03-20:
Not sure this is something new, when doing FE changes i've converted most white spaces/indentation to adhere to https://github.com/OpenStickCommunity/GP2040-CE/blob/main/.editorconfig
using prettier, unsure what other devs are using.

We added .gitattributes for line endings but this should not affect white spaces/indentation.
https://github.com/OpenStickCommunity/GP2040-CE/blob/main/.gitattributes

I've seen some (old) files that are using mixed like: https://github.com/OpenStickCommunity/GP2040-CE/blob/main/www/src/Pages/SettingsPage.jsx

At least for the web config we could have a linter as build step, been hesitant to add that thought.

bsstephan, 2024-03-22:
I dropped the link yesterday but I would recommend having a `./.clang-format` and having VScode, vim, etc. use that to enforce linting. Seems like the best common tool I've seen around.

arntsonl, 2024-06-03:
I'm going to close this as this isn't something we can just "enable" in github

closed 2024-06-03 by arntsonl