OpenStickCommunity/GP2040-CE #902: How to compile the version of haute42 of 0.7.7?

okabesg, 2024-03-20:
### Expected Behavior

I expect to get a GP2040-CE_0.7.7_haute42.uf2 file

### Current Behavior

I clone the repo and switch to the Haute42 branch, which called 20240115-Haute42-update. After that, I create the build dir and run 
`cmake ..` 
and
`make`.
But all I got was GP2040-CE_0.7.6_Pico.uf2 file, which is the wrong version.
How to fix this? Thanks a lot!

### Screenshots & Files

[image]

[image]

Pelsin, 2024-03-20:
This is working as intended and is not a bug.
Please read through the documentation for firmware development: https://gp2040-ce.info/development/firmware-development and the documentation in general for these types of questions.

The default board config to use is pico: https://gp2040-ce.info/development/firmware-development#building

A haute config also exists in main or an artifact in the latest release, would be better than using a stale branch.

If you have question, you are welcome to join the discord.

okabesg, 2024-03-20:
The documentation works for me! Thank you so much!
I'm new here and didn't know how to find useful infomation in the documentation.
Thanks again!

closed 2024-03-20 by okabesg