OpenStickCommunity/GP2040-CE #905: Haute42 T13 often fails to connect to Xbox Series X

CharliePRamsey, 2024-03-22:
### Expected Behavior

When the input mode is set to Xbox One, Xbox One Passthrough is enabled, and a Mayflash Magic-X is plugged into the right-hand port, the device should consistently be functional on an Xbox Series X.

### Current Behavior

When plugged into the Xbox, the T13 will seem to attempt to authenticate with the console for a few seconds (variable), before eventually freezing up. The LED's on the buttons and the button display on the top left of the controller will stop responding to input. If buttons were being pressed when the freeze occurs, those LEDs and buttons on the display will remain lit forever. The Magic-X will continue flashing alternating blue, yellow, and green lights. Sometimes the Magic-X will just flash blue. In all success cases, it flashes blue, but it will sometimes flash blue in a failure case as well. 

This is never resolved without unplugging the device. Rarely, plugging in the controller will work as normal.

If the T13 is plugged into the Series X while the console is booting up from a hard reset, it will work on that first attempt with 100% consistency. Subsequent attempts while the Xbox is turned on will go back to the above behavior. Note that it isn't just that the first attempt after a reset works, it has to be _plugged in_ while the console is booting up.

The controller works as expected when using firmware 0.7.6.

### Context

* Firmware version: 0.7.7
* Name of device: Hautepad T13
* Link to where I purchased the device: https://www.aliexpress.us/item/3256806057935826.html?spm=a2g0o.order_detail.order_detail_item.4.45eaf19cwlDjs5&gatewayAdapt=glo2usa

* Mayflash Magic-X Firmware Version: 1.31

Note also that the Series X has access to Dev Mode, although it was in Retail when observing the above behavior.

### Steps to Reproduce

Step 1. Update the T13 to firmware 0.7.7
Step 2. Update the Mayflash Magic-X to firmware 1.31
Step 3. Plug Magic-X into right-hand port of T13.
Step 4. While the Xbox Series X is turned on, plug the controller into the front-facing USB port of the console.
Step 5. Observe behavior.

### Screenshots & Files

During one of my attempts, one of the button LEDs froze in an orange state. They usually lit up white when pressed, so was unsure if this meant something.

[image]

TheTrainGoes, 2024-03-22:
Hi @CharliePRamsey ,

Can you please link the v0.7.7 you put on the unit and confirm if you did a nuke when you updated to v0.7.7?

Re: "the device should consistently be functional on an Xbox Series X." we have not observed any issues with the Magic-X however if it cycling colors then it means the Magic-X is changing modes.

Re: "Note also that the Series X has access to Dev Mode" This should have no impact on things unless you have also modified the Xbox.

CharliePRamsey, 2024-03-22:
Link for v.0.7.7 - https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.7/GP2040-CE_0.7.7_Haute42.uf2

I did a flash nuke.

TheTrainGoes, 2024-03-22:
Can you please post a video showing how its not working?

CharliePRamsey, 2024-03-23:
https://github.com/OpenStickCommunity/GP2040-CE/assets/26028422/324499ce-fcd2-4101-870e-cfcc3cc34d43

spicyspaghetti, 2024-03-25:
hey  @CharliePRamsey , I have haute board mini ,an xbox like yours and am a tekken player like you
I was searching for some updates and noticed you mentioned an haute board working on xbox , I was like it impossible until I saw you using the Mayflash Magic-X , first time I heard of it.
Can you tell me if Mayflash Magic-X work on my haute board mini because I wanna buy one and am not sure, and If you can add me on xbox my gamertag is Drex699669

TheTrainGoes, 2024-03-25:
Hey @spicyspaghetti , I updated your comment please don't put your Gamertag in here.

There are two versions of the Haute42 Mini, one with and one without the USB passthrough port.  If you get the one with the USB passthrough port there are no issues using it on Xbox with an appropriate dongle.  

I have tested it with both the Mayflash Magicboots Xbox and the Mayflash Magic-C.

TheTrainGoes, 2024-03-25:
> Forgive the quality / angle
> 
>  VID_20240322_220952.-.Trim2.mp4

Few things regarding this:

1 - If the Xbox is turned on and previously had issues connecting to the controller it may get locked up until it is power cycled (proper power cycle not just a reset).
2- It can take 15-20 seconds for the auth to take place and buttons should not be pressed during this time

I have sent you a copy of the beta build that should address the issue, you said there were issues with the Home button - Can you please elaborate on those.

spicyspaghetti, 2024-03-25:
Thank you for replying @TheTrainGoes , sorry about the Gamertag.

I really appreciate your response and I do own hauteboard mini with the usb passthrough port, always wondered what that was for.

Thanks again and have a nice day.

strusic, 2024-03-27:
I have same problems on my diy flatbox rev5 board, but I thought it was due to poor connections on board. When RC of 0.7.8 will be published to tests?

TheTrainGoes, 2024-03-27:
@strusic We will have it posted as an RC when ready.  Until then you can build from main if you like to test our current build.

cmattinson, 2024-04-04:
I was experiencing the same issue and just updated my Haute42 M16 to v0.7.8-RC1 using Magic-X 1.3.1, I can confirm it has resolved my connection issues with Xbox Series X. The controller now consistently connects after the console is woken from standby.

TheTrainGoes, 2024-04-15:
This issue has been addressed as of v0.7.8.

If there are additional issues please submit a ticket ticket.