OpenStickCommunity/GP2040-CE #924: PS Touchpad coordinates on button presses as actions

mikepparks, 2024-03-28:
Currently, our touchpad implementation sends the button state when pressed, but does not send any positioning data, opting to set the state as untouched.  It's unknown whether this would cause undesired behavior in some games, but for current uses, there may be value including some sort of positioning data when the button is pressed.  A known implementation of this defines regions for the position coordinates based on the state of the D-Pad, but that could pose some additional behavior defects.  For example:

* D-Pad Neutral + TP = Center of Touchpad
* D-Pad Left + TP = Left of Touchpad
* D-Pad Right + TP = Right of Touchpad

A proposed means of implementing this would be to set different "actions" that could be bound to button presses instead, avoiding the need for a hotkey or macro dependency.

* Neutral - Current Touchpad behavior
* Left
* Right
* Center

JayJTD, 2024-05-12:
Hey, i'd like to also request this enhancement if possible. 

With the current implementation, PS1 and PS2 Classics on PS4 are completely unplayable with a GP2040-CE controller. Sony's implementation of the PS1 and PS2 classics require both a Left and a Right Touchpad input instead of the neutral one that is currently used. Left Touchpad is the "Select" button while Right Touchpad is the "Start" button. 

The PS1 and PS2 Classics versions of Tekken 1, Tekken 2, and The King of Fighters 2000 are not able to be started at all due to their requirement of pressing Start instead of the usual pressing of X/Cross.

A feature like this will allow the GP2040-CE to become even closer to feature parity with the Brooks UFB.

Brunoleog, 2024-06-14:
I have the same problem....I set up an arcade with a ps4 and ps2 games that require "Start" to start cannot be played because the start is on the right touch pad.

Brunoleog, 2024-06-14:
I have the same problem....I set up an arcade with a ps4 and ps2 games that require "Start" to start cannot be played because the start is on the right touch pad

bsstephan, 2024-06-14:
I have the makings of an implementation, so I'm just going to assign this to myself before I forget.

Brunoleog, 2024-06-14:
Thank you very much, send me news about the implementation.

commit b759c638 (2024-06-17): use new A3 and A4 to emulate tpad click position on PS4 (#1060)

* use touchpad + dpad to emulate tpad click position on PS4

when the TP "button" is pressed, we formerly did nothing with the
touchpad data, including not even sending the bit to indicate there is
touchpad data, which was fine for normal purposes. however, we can use
the dpad to emulate position data as if a finger was actually touching
it, and send touchpad data for the eight directions. this will allow us
to simulate clicking the left vs. right sides of the touchpad, which
will satisfy the PS1-on-PS4 emulation of the Start and [... cut]
closed 2024-06-17 by arntsonl