OpenStickCommunity/GP2040-CE #928: Keyboard keys are not remamppable after PR #890

octopuserectus, 2024-04-02:
### Expected Behavior

Remapping a key in Keyboard Mapping should take an effect
### Current Behavior

Remapping a key takes no effect after PR #890 

### Context

* Firmware version: v0.7.8-RC1
* Board config name: Haute 42
* Name of device: Haute Pad S16
* Link to where you purchased the device https://aliexpress.com/item/1005006421550898.html

### Steps to Reproduce

1. Boot into web-config
2. Go to /settings
3. Choose keyboard from the dropdown
4. Change any key to a non-default value
5. Save changes
6. Reboot into hid-kb mode
7. Press the key we just changed---it's unchanged

arntsonl, 2024-04-02:
Cool ty, we'll look into it. I'm assuming I missed something on the www react code when I ported it over to that page.

octopuserectus, 2024-04-10:
Closing since it's fixed with #947

closed 2024-04-10 by octopuserectus