OpenStickCommunity/GP2040-CE #930: 7.8 RC1 Macro bug

wesleysni, 2024-04-02:
### Expected Behavior

> When using 
[image] 
Macro Activation Type : Hold Repeat

Prior to upgrade to 7.8 It is supposed to run the macro while pin is held down, then stop when released. 

### Current Behavior

When activating macro pin, macro repeats but does not stop when releasing macro pin. (Worked properly in 7.7)

### Context

Version
RP2040 Advanced Breakout Board (GP2040-CE_0.7.8_RP2040AdvancedBreakoutBoard.uf2)

Current: v0.7.8-RC1

Latest: v0.7.7

Memory (KB)
Flash: 913.87 / 2048 (44.62%)
Heap: 14.69 / 152.09 (9.66%)

Static Allocations: 111.91

os: windows 10
webbrowser: google chrome

### Steps to Reproduce

I upgraded from 7.7 to 7.8
Macros and all other settings were programmed on 7.7 and carried over to 7.8 after upgrade
After upgrade I tested this in Street Fighter 6 
Activate macro and it repeats continuously without being able to stop (until i unplug controller)
I will provide pictures below

### Screenshots & Files

[image]
[image]
[image]
[image]

arntsonl, 2024-04-02:
Perfect, thank you for all of the details! We'll get this replicated and figured out.

wesleysni, 2024-04-02:
I also nuked then flashed to 7.8 and problem still persists as well.

arntsonl, 2024-04-06:
Should be fixed in RC2. Will verify on full release and close.

TheTrainGoes, 2024-04-15:
This has been addressed in release + hot fix

closed 2024-04-15 by TheTrainGoes