OpenStickCommunity/GP2040-CE #934: WebConfig -  If B1 is pressed 10 times will cause the display to freeze and causes webconfig to become non-responsive

InfraredAces, 2024-04-03:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm your issue exists on the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Expected Behavior

Be able to press B1 any number of times to preview the layout and then switch back.

### Current Behavior

If B1 is pressed 10 times will cause the display to freeze and causes webconfig to become non-responsive. B2 when pressed multiple time does not result in this behavior.

### Context

> Please provide all relevant information about your setup. This is important in case the issue is not reproducible
> except under certain conditions. Important elements include:
>
> * Firmware version: 0.7.8 RC1
> * Board config name (also from webconfig, please)
> * Name of device
> * Link to where you purchased the device
>
> If your issue is with the web configurator, also include:
>
> * Operating System
> * Browser

- 0.7.8 RRC1
- Haute42
- Haute42 S16
- https://www.aliexpress.us/item/3256806235236146.html
- 6.8.1-arch1-1 (64-bit)
- Chromium

### Steps to Reproduce

1. Flash Nuke board
2. Flash board with [0.7.8 RC1](https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.8-RC1/GP2040-CE_0.7.8_Haute42.uf2)
3. Boot into web config
4. Press B1 10 times

### Screenshots & Files

> Please include any relevant screenshots, photos/recordings, or files here.

When booting into Webconfig
[image]

Web Config once B1 has been pressed 10 times and web config is refreshed
[image]

Page after a minute of cycling
[image]

arntsonl, 2024-04-06:
Should be fixed in RC2. Will verify on full release and close.

mikepparks, 2024-04-10:
Should not be crashing in v0.7.8.

closed 2024-04-10 by mikepparks