OpenStickCommunity/GP2040-CE #945: The400 mini - is a configuration possible?

lininjim2, 2024-04-06:
I've got the new mini console emulating the Atari 400 (and up to 130XE). The supplied stick isn't great so I got some picos to wire up a D9 connector to use original controllers but I'm not getting anywhere. Would a configuration for this mini be possible please?
They maker provides config info that sadly I'm unable to interprete.

Thanks

https://retrogames.biz/support/the400-mini/controller-configuration/

bsstephan, 2024-04-07:
Given that this device allows you to configure mappings for any USB HID device, I suggest you do that and that this request is consequently outside of the scope of our project.

It looks like there are some tools that will help you generate the config: https://github.com/mdqinc/SDL_GameControllerDB?tab=readme-ov-file#mapping-tools

closed 2024-04-07 by bsstephan

lininjim2, 2024-04-08:
I appreciate the update, I had tried that and gotten nowhere, I was asking as there are other Mini consoles in the settings. Cheers anyway.

bsstephan, 2024-04-08:
The existing mini console modes exist because those consoles do not have the capability of being configured to bring your own device, as yours does. Yours is more open and I recommend you utilize that openness.

lininjim2, 2024-04-08:
I tried, I failed. Again thanks for the reply though.