OpenStickCommunity/GP2040-CE #951: Hope to add Left/Right Priority SOCD mode

JJJY5, 2024-04-09:
###Problem

When using a normal shoryoken(→↓↘)character in Neutral SOCD, →↓↑ is a very good way to simplify the command. In this situation, ↓↑ is neutral, this two commands are equal.
In Last Win SOCD, ↓↑ is not neutral but ↑. Althought it still works, it requires the ↑ and the P or K command to be pressed at the same time. If not, the character will jump and the shoryoken will fail to antiair.
But Last Win SOCD is very useful at footsies. It allows you to use any commands of shoryoken without releasing ←.
What If there is a priority socd only for left and right? Then we can do this without releasing ← at footsies and no need to worry the character will jump when you actually want to use shoryoken by inputing →↓↑

### Solution

Add Left/Right Priority SOCD mode, which could be made by banning the Up/Down's function of Last Win SOCD mode.

### Do you plan to work on this feature yourself?

No, I am just a normal player, only know a little bit about python.

JJJY5, 2024-04-11:
@TheTrainGoes That’s exactly what I mean, hoping to add a last win socd mode exclusively for left/right.

TheTrainGoes, 2024-04-11:
We have talked about adding a `per axis` configuration for more advanced SOCD but its very situational.  For now we are happy with the current SOCD offerings but will consider this in the future.

JJJY5, 2024-04-11:
@TheTrainGoes 
I find this function in Cangyan Chip 2.0+, which is a quite famous fighting arcade chip before GP2040 came into being. I hope this could be add to GP2040 as well. In fact, Up Prioity is also situational, but more is always better less. Thanks for your contribution.

closed 2024-04-11 by JJJY5