OpenStickCommunity/GP2040-CE #958: Some screens do not work in 0.78 version

wuxinygq, 2024-04-11:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm your issue exists on the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Expected Behavior

> Please describe the behaviour you are expecting.

### Current Behavior

> What is the current behaviour? If this is something that worked in the past, be sure to include the conditions in
> which it did (prior version of the firmware, a different GP2040-CE device where it works, etc.).

### Context

> Please provide all relevant information about your setup. This is important in case the issue is not reproducible
> except under certain conditions. Important elements include:
>
> * Firmware version (please get the full version listed in webconfig, if at all possible)
> * Board config name (also from webconfig, please)
> * Name of device
> * Link to where you purchased the device
>
> If your issue is with the web configurator, also include:
>
> * Operating System
> * Browser

### Steps to Reproduce

> Please provide detailed steps for reproducing the issue.
>
> 1. Step 1
> 2. Step 2
> 3. ETC...

### Screenshots & Files

> Please include any relevant screenshots, photos/recordings, or files here.

mikepparks, 2024-04-11:
@wuxinygq Please read the form and fill in the issue with more detail. Stating "some screens do not work" with no details gives us nothing to work with.

Garouuuu, 2024-04-11:
> @wuxinygq请阅读表格并更详细地填写问题。陈述"某些屏幕不工作"而不提供任何细节,这让我们无从下手。

sh1106  i2c0  0x3c
SDA-0  SCL-1

Garouuuu, 2024-04-11:
@mikepparks

TheTrainGoes, 2024-04-11:
This is a known issue @wuxinygq.

We will look to resolve in v0.7.9

TheOpponent, 2024-04-14:
My SH1106 screen is also not working on Waveshare RP2040 Zero on 0.7.8 mapped to i2c1, where it did work in 0.7.6.

mikepparks, 2024-04-17:
Please use one of the test builds in PR #974 and comment on this issue with the type of display you're using, in addition to the information the screen displays.

TheOpponent, 2024-04-17:
> Please use one of the test builds in PR #974 and comment on this issue with the type of display you're using, in addition to the information the screen displays.

Using v0.7.8-5-g6078ece for Waveshare Zero from #974, my SH1106 display on i2c1 reads:
```
Screen Type: 4
Screen Reg: 8
Detected Disp: 2
```

mikepparks, 2024-04-24:
With the latest update to PR #974 the displays should be back in working order.  We'll get it merged in for 0.7.9 once we have more confirmations that things are working.

TheOpponent, 2024-04-24:
Screen features in the most recent artifact for Waveshare RP2040-Zero now works as expected.

TheTrainGoes, 2024-06-03:
This has been addressed as of #974 - no reply from original poster.

closed 2024-06-03 by TheTrainGoes