OpenStickCommunity/GP2040-CE #960: Macro frames bugs

rebornsick, 2024-04-11:
> Please read these quoted sections before submitting, and remove them (including this one) when you have provided what
> is asked, in order to make addressing the issue easier for everyone. Thank you!
>
> Be sure to:
> * Check the documentation [HERE](https://gp2040-ce.info/) to answer many questions on firmware capabilities and usage.
> * Make sure that your issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues).
> * Confirm your issue exists on the latest version of the firmware, found
  [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases).

### Expected Behavior

Each action in the settings lasts the same number of frames

### Current Behavior

In Street Fighter 6

- frames for every action doesn't work
- after reboot device many times for testing macro settings, can't save in macros configuration

### Context

- Flashed nuke firmware
- flash latest 0.7.8.pico firmware
- disable turbo add-on
- setup maros
- change pin mapping

### Steps to Reproduce

### Screenshots & Files

[image]
[image]

TheTrainGoes, 2024-04-11:
Hi @rebornsick ,

To ensure I am understanding this correctly, you are expecting that "Each action in the settings lasts the same number of frames" despite it being listed as different numbers of frames in the attached screenshot?

Have you attempted to do this via MS settings and play around with those at all?

rebornsick, 2024-04-11:
Hi @TheTrainGoes ,

I mean no matter what number of frames I set, macros only take 1 frame for every action.

I tested both 0.7.7 and 0.7.8 with only changing macro configs and mapping the same macro button, 0.7.7 works as I expected and 0.7.8 only takes 1 frame as you see. (1, 2, 3, 4, 5 frames for 5 actions)

[image]

TheTrainGoes, 2024-04-11:
@rebornsick are you able to replicate this in MS mode (opposed to frames)?

rebornsick, 2024-04-11:
@TheTrainGoes MS means millisecond？yeah it's same.

TheTrainGoes, 2024-04-11:
@rebornsick so you changed it to MS, made adjustments and the firmware is not respecting those adjustments?

rebornsick, 2024-04-11:
@TheTrainGoes yes, it's same, only 1 frame. So you can't replicate this? does this relate to hardware? I'm using haute42 S16 but with pico firmware though

rebornsick, 2024-04-11:
@TheTrainGoes OK, this doesn't affect me much, just letting you know.Thanks for your work 🫡

rebornsick, 2024-04-13:
@TheTrainGoes Hello, I just noticed that we re-compiled 0.7.8 firmware and it fixed this frame bug.

TheTrainGoes, 2024-04-15:
Hi @rebornsick ,

Yes, this is addressed now and I will close the ticket off.

closed 2024-04-15 by TheTrainGoes