OpenStickCommunity/GP2040-CE #961: LCD timeout setting doesn't work in 7.8

hctarks, 2024-04-11:
### Expected Behavior

> Set LCD timeout on Haute42 device to 1 minute. LCD screen should turn off by itself after one minute.

### Current Behavior

> LCD screen does not turn off after 1 minute.

### Context

> Please provide all relevant information about your setup. This is important in case the issue is not reproducible
> except under certain conditions. Important elements include:
>
> * Haute42 7.8.

JJJY5, 2024-04-13:
I met the same problem in another device with a raspberry pico chip.

TheTrainGoes, 2024-04-21:
This is a known issue due to the display re-write that happened in v0.7.8.  We will look to fix it for v0.7.9.

mikepparks, 2024-04-24:
Please verify that this is working as expected in PR #974

TheTrainGoes, 2024-06-03:
This has been addressed via #974 - no reply from original poster.

closed 2024-06-03 by TheTrainGoes